LinuxCNC/linuxcnc #705: building a debian package fails on current ubuntu 19.10

bruns01, 2020-03-24:
## Here are the steps I follow to reproduce the issue:

 1. install a fresh ubuntu 19.10
 2. Clone linuxCNC repository (master) and follow build instructions
 3. add missing dependencies until "building as run in place" succeeds
 4. follow instructions to "build a debian package"

## This is what I expected to happen:
successful build with a linuxcnc.deb file as a result

## This is what happened instead:
virtually unlimited error messages from dblatex :

Cannot convert 'common/images/emc2-intro.png' to pdf
Cannot convert 'user/images/whatstep1.png' to pdf
Cannot convert 'gui/images/axis.png' to pdf
Cannot convert 'gui/images/touchy.png' to pdf
...
and a fail of the build in the documentation section. 

## What i found out
The problem seems to be the dependency of dblatex on inkscape for graphics conversion to 
pdf-format. 
While svg -> pdf succeeds with "inkscape -z -D --pdf-export=outputfilename.pdf inputfile.svg"
the same call to inkscape with "inputfile.png" crashes because inkscape wants to open a GUI-Window to ask several questions like "embed or link graphics", "what resolution has the png", etc ...

According to the inkscape manual there is no CLI-option to avoid such GUI interaction on png-import.

This is a show stopper with the current inkscape versions.

## Information about my hardware and software:

 * I am using : Ubuntu 19.10 
 * I am using this kernel version: 5.3.0-42-lowlatency #34-Ubuntu SMP PREEMPT Fri Feb 28 06:31:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
 * I am trying to test linuxCNC on a current Ubuntu-OS
 * I am using this LinuxCNC version 2.9.0~pre0 (freshly cloned from git master)

andypugh, 2020-03-24:
I saw a similar string of messages at the weekend, but I have just built a set of debs now with no (significant) problems on Debian 10 (Buster)
Have you tried dpkg-checkbuilddeps ? I have a vague memory that (despite appearances) it was a missing dependency.

bruns01, 2020-03-25:
Yes, I did dpkg-checkbuilddeps and, in fact, had to install several dependencies even beyond what that told me. As I wrote the "run in place build" succeeded.

I tried the respective inkscape cli command in the shell 
   inkscape -z -D --export-pdf=test.pdf  linuxcnc/docs/src/common/images/emc2-intro.png
and it stoped with a modal dialog popping up, that cannot be resolved in an automated build.

Do you know, where to find the dblatex configs, that (possibly) control the call to inkscape?

andypugh, 2020-03-25:
I don't see the errors I saw before. I see different errors, but still get the debs. I don't know what actually changed. 
(Checks) Well, apart from being in 2.8 rather than master. 

This is the output with 2.8, I will try master. 

inkscape -z -D --export-pdf=fig42.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/eight-holes_en.svg
inkscape -z -D --export-pdf=fig43.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/twelve-holes_en.svg
inkscape -z -D --export-pdf=fig44.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/g81mult_en.svg
inkscape -z -D --export-pdf=fig45.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/g81ex1_en.svg
inkscape -z -D --export-pdf=fig46.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/g81ex2_en.svg
inkscape -z -D --export-pdf=fig47.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/g81_en.svg
inkscape -z -D --export-pdf=fig48.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/g81a_en.svg
Cannot convert 'gui/images/image-to-gcode.png' to pdf
Cannot convert 'gui/images/i2g-roughing.png' to pdf
Cannot convert 'gcode/images/ToolTable-TouchOff.png' to pdf
inkscape -z -D --export-pdf=fig49.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/comp-path_en.svg
inkscape -z -D --export-pdf=fig50.pdf /home/andypugh/linuxcnc-dev/docs/src/gcode/images/comp02_en.svg
Cannot convert 'gcode/images/outside-comp.png' to pdf
Cannot convert 'gcode/images/inside-comp.png' to pdf
Cannot convert 'gui/images/tooledit.png' to pdf
Cannot convert 'gui/images/tooledit-sort.png' to pdf
Cannot convert 'gui/images/tooledit-columns.png' to pdf
Cannot convert 'config/images/servo-feedback.png' to pdf
Cannot convert 'config/images/pid-feedback.png' to pdf
Cannot convert 'config/images/latency-test_en.png' to pdf
Cannot convert 'config/images/latency-plot.png' to pdf
Cannot convert 'config/images/latency-histogram.png' to pdf
Image 'images/configuration-selector.png' not found

andypugh, 2020-03-25:
Output is the same with master, get a set of 2.9 .deb files, including docs. 

andypugh@rm-one:~/linuxcnc-dev$ inkscape --version
Inkscape 0.92.4 (5da689c313, 2019-01-14)

Is that the difference?

bruns01, 2020-03-25:
Can't check the version right now, I'm not close to the machine.
According to pakages.ubuntu.com inkscape 0.92.4-4 should be installed, hence it's probably not the reason/difference. 
I have to re-check the errors on my build, I wonder how it could run successful for you without the proper graphics available. 
Maybe I misinterpreted the root cause of the fail of my build.
I'll be back later.

closed 2020-03-26 by bruns01

phillc54, 2020-03-26:
I just noted in your original report that uname reported: 5.3.0-42-lowlatency #34-Ubuntu SMP PREEMPT Fri Feb 28 06:31:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
You need either a **RTAI** or **PREEMPT RT** kernel

bruns01, 2020-03-26:
Thanks for the confirmation. I realised that in the meantime and will try to build a custom RT-kernel for Ubuntu as a next step.
I found it not so easy for a RT- beginner to find current, reliable information about the subtle differences of RT, RTAI, Xenomai, low latency, ... and the like. (I'm working with Linux since kernel version 0.98 on 3.5" floppy-disks but never with RT so far).

phillc54, 2020-03-26:
You may be able to use a prebuilt PREEMPT RT image from the Debian packages.

bruns01, 2020-03-26:
Thanks, a brilliant idea, that may save me a lot of time and frustration!

phillc54, 2020-03-26:
I forgot to mention that if you want RTAI then there is a package at linuxcnc.org/temp that Andy has built