LinuxCNC/linuxcnc #707: G72 fix

papagno-source, 2020-04-02:
1.start lathe.ini
 2.load this file 
 3.G54 G18 G21 F400
G40 T2M6G43
m3 s100
o100 sub
g1 x50 z-40f10000
x100 z-39.999
o100 end sub
G72 Q100 X50 Z0 I3 R0.1
g0 x0 z0
m30

This work ok , but if change in 

G54 G18 G21 F400
G40 T2M6G43
m3 s100
o100 sub
g1 x50 z-40f10000
x100 z-40
o100 end sub
G72 Q100 X50 Z0 I3 R0.1
g0 x0 z0
m30

This not is drawed, because ?

## It worked properly before this:
Actual Master 2.9.pre0

## Information about my hardware and software:

 System Debian Wheezy

andypugh, 2023-10-04:
I have made some progress with this, but not a great deal. 

The "magic" is all in src/emc/rs274ngc/interpg7x.cc

Modifying the area around line 765:

````

    while(p!=end()) {
	printf("765 x = %f \n", x);
	while(x-tolerance>imag(location))
	    x-=delta;
	printf("x = %f\n", x);
	if((*p)->dive(location,x,out,p==begin())) {
	    	printf("dive dive dive\n");
	    if(cycle==1) {
		/* After the initial roughing pass, move along the final
		   contour and we're done
````

I see the "While" execute once for each line in the cycle for the "good" file, then "dive, dive, dive" whereas the bad file runs once and straight to dive. 

I haven't figured out yet whether this is due to the p being shorter or the dive() function returning a different result. 
I really know very little about C++ and this is using a lot of iterators and complex numbers. 

Good

[image]

Bad

[image]

andypugh, 2023-10-04:
It seems that imag(back()->ep) gets flipped somewhere before do_g71 the value on entry is +40 and there is no +40 in the G-code. 

If the condition below in the "swap" routine  is changed to "if (dir_x>=0)" then both test cases pass. 

_but_ then the code fails with z = -40.00001

````
    /* Change the direction of the profile to have Z and X decreasing */
    void swap(void) {
	double dir_x=imag(front()->ep()-back()->ep());
	double dir_z=real(front()->ep()-back()->ep());
	if(dir_x>0) {
	    for(auto p=begin(); p!=end(); p++)
		(*p)->flip_imag();
	    flip_state^=2;
	}
	if(dir_z<0) {
	    for(auto p=begin(); p!=end(); p++)
		(*p)->flip_real();
	    flip_state^=1;
	}
    }
````

andypugh, 2023-10-06:
Thanks mark-v-d. Seems to be fixed.

PR #2677 (2023-10-06): Corrected G72 behaviour with no axis movement.
Fixes #707.
commit 03793cc9 (2023-10-06): Corrected G72 behaviour with no axis movement.

Fixes #707
closed 2023-10-06 by andypugh