LinuxCNC/linuxcnc #709: Single step mode skips entire loop contents in some circumstances.

andypugh, 2020-04-05:
## Here are the steps I follow to reproduce the issue:

1. Open sim/axis-mm
2. Home to X0 Y0
3. Run this .ngc file starting it with the single-step button.

`#1 = 0
O100 repeat [6]
(DEBUG, 1)
G1 F50 X[5 + #1]
(DEBUG, 2)
G1 F50 Y[5 + #1]
(DEBUG, 3)
G1 F50 X[10 + #1]
#1 = [#1 + 5]
O100 endrepeat
M2`

## This is what I expected to happen:

The program to run step-by-step

## This is what happened instead:

After the first couple of single-steps the machine runs straight to the end-point, ignoring the entire loop contents. 

## It worked properly before this:

Prior to https://github.com/LinuxCNC/linuxcnc/pull/151

The issue is caused by a particular set of circumstances and the code here:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/emctaskmain.cc#L572
`  if ( programStartLine != 0 &&
				 emcTaskPlanLevel() == 0 &&
				 ( programStartLine < 0 ||
				   emcTaskPlanLine() <= programStartLine )) {
				// we're stepping over lines, so check them
				// for
				// limits, etc. and clear then out`

In single-step mode programStartLine == 1 throughout. 
On getting to the loop, because the loop is on line 1 (counting from zero) the conditions are satisfied, and the system steps-over the loop. 
The program runs normally with a spacer line of code above the O100

zultron, 2020-05-11:
I'm starting on this today.

zultron, 2020-05-12:
Fix in PR #852.  Thanks again, @andypugh!

closed 2020-05-12 by andypugh