LinuxCNC/linuxcnc #712: Make G-code access to HAL and INI the default

andypugh, 2020-04-07:
Note to self to change the defaults in 
http://linuxcnc.org/docs/2.7/html/remap/remap.html#_optional_interpreter_features_ini_file_configuration

(And probably replace them with specific INI file entries)

andypugh, 2020-04-10:
Work in progress https://github.com/LinuxCNC/linuxcnc/tree/andypugh/persistentG43
6999b04d5b48a854a6c3325822ae89aa61ca7a0b

Two things: 
I don't see persistent G43 having the expected effect. 
It seems that the ini scanning tools in rs274ngc_pre.cc only work for 0 and 1 booleans, whereas the code very much seems to expect true / yes / 1 to all return 1. 
https://github.com/LinuxCNC/linuxcnc/blob/andypugh/persistentG43/src/emc/rs274ngc/rs274ngc_pre.cc#L878

rene-dev, 2020-04-12:
I just had a look at various manuals(fanuc, haas, DIN), and it seems that I was wrong.
The current g43 behaviour is the same, you need to do it again after every tool change.
The retain flag does work, but maybe you just discovered another bug...
I was only looking at the length displayed in the Botton of axis, and the tlo displayed in the dry.
Probably easier to keep it the way it is, as many tests break by changing the default.

andypugh, 2020-04-12:
OK, I will default to off. It didn't seem to be a popular feature on the mailing list. 
When you say that the flag works, do you mean just that the flag is set, or that G43 gets re-applied?

rene-dev, 2020-04-13:
G43 gets reapplied.

andypugh, 2020-04-13:
Let's take this to email and compare notes.

andypugh, 2020-04-18:
Completed as part of 74fc4491014b9de4141c44f58e391960e681343b

closed 2020-04-18 by andypugh