LinuxCNC/linuxcnc #715: huanyang-vfd incorrect byte order when writing to 8 bit registers (e.g. PD143).

dlushni, 2020-04-08:
https://github.com/LinuxCNC/linuxcnc/blob/56cedd4d36ef761325f1e32ec36873f9ea55e08f/src/hal/user_comps/huanyang-vfd/hy_vfd.c#L472

Huanyang VFD has some PDxxx registers that are 2 bytes in length (e.g. PD004, PD005, PD142, etc. all are), and some register that are 1 byte in length (e.g. PD143). At this moment, PD143 is the only 8-bit register I found referenced in huanyang-vfd code.

Packet format for FUNCTION_WRITE request is different for writing to 1 byte and 2 byte registers. My experiments showed that sending 1 byte to a 2-byte register results in invalid operation and register contents are unchanged.  Sending 2 bytes to a 1-byte register is accepted and modifies the register. However, the leftmost byte of data is the value that's actually being written to the register. This corresponds to little-endian byte order (where MSB is ignored). Also this is the same byte position in the packet as if 1 byte was sent to a 1-byte register.

This is different than the byte order used for writing 2 bytes to a 2-byte register, and what is actually implemented in the code:
https://github.com/LinuxCNC/linuxcnc/blob/56cedd4d36ef761325f1e32ec36873f9ea55e08f/src/hal/user_comps/huanyang-vfd/hy_comm.c#L499
Here the leftmost byte contains MSB bits of the 16-bit value, which corresponds to big-endian format.

Looks like the line of code that writes "motor_poles" value to PD143 is incorrect and needs to be fixed (e.g. by shifting bits [7..0] to position [15..8]).
https://github.com/LinuxCNC/linuxcnc/blob/56cedd4d36ef761325f1e32ec36873f9ea55e08f/src/hal/user_comps/huanyang-vfd/hy_vfd.c#L472

andypugh, 2020-04-09:
I have tried making the change:
            hc_data->data = (uint16_t)(*haldata->motor_poles) << 8;

But nothing seems to change as far as operation of the VFD. 

My lathe has a 4-pole motor. I added -P 8 to the loadusr hy_vfd line and the motor runs at the correct speed (I believe with the wrong pole-count it should run twice as fast) 

I made the change above, recompiled, and the motor speed remained correct. 

With and without the change the HAL pin that reports motor poles shows 8. 

Is there something wrong with my test? I can't easily access my HY VFD without pulling my lathe away from the wall.

andypugh, 2020-04-24:
Any update? I need a way to see the problem in order to fix it.

petterreinholdtsen, 2023-10-03:
@dlushni Any hope you will be able assist in reproducing the issue?