LinuxCNC/linuxcnc #725: Spindle Orient V2 speaking.

alkabal, 2020-04-12:
Hi

This is not a issue but a i like to discuss about a second orient component revisited.

I hope what i have done is good on the logic fondamental but i'm pretty sure the speed management/positioning can be really better with some more advanced idea. (off course i can understand if this is not really good, not interesting/wanted or other.)

-For now this component try to include all needed input/output for allow a more easy hal configuration (no more need or2, and2 scale etc).
-Added spindle autohoming multi mode : possible with machine homing or at first M3/M4/M19
-Also added a jog mode for Mpg wheel

All in one file with exemple hal inside.)
[orientv2.comp.txt](https://github.com/LinuxCNC/linuxcnc/files/4467327/orientv2.comp.txt)

alkabal, 2020-04-15:
Hi

Well, saddly there is not too much person that like to play with orient.

Maybee later so i like to push here another attempt, i hope this is better like that i have somme good result with my spindle encoder resolution of 0.9° per count.

Nothing change regarding to the homing process, but completly rewrited move sequence for try to match and adjust with ~0.5° precision. For now Mode 1 and 2 are not really great but work, mode 0 and mode jog seems to work something like good. (sometime the position is too close from one to other and some little but fast oscillation does not allow to be is_oriented = 1)

I really hope some review.

Br
[orientv2.comp.txt](https://github.com/LinuxCNC/linuxcnc/files/4478434/orientv2.comp.txt)

andypugh, 2020-04-18:
Minor niggles: Some of the pin names sound odd to me: I think that angle-asked-out for example should be angle-command-out. Your component is too polite:
Your } else { indenting is rather unconventional. LinuxCNC does not have as consistent a coding style as might be desired, but it would be nice to be at least vaguely consistent. 
http://linuxcnc.org/docs/html/code/style-guide.html

alkabal, 2020-04-18:
Hi

Thanks a lot for message, i will try to update coding style you are right. And ok i can rename some pin soon.

Rgarding to logic moving / stopping / Homing / jog inside code is the most wanted ^^ 

Also i like to know if this is interesting for all or continuing this only for my own use  ? (i does not really need orient for my lathe and mill, this is mostly for brain torture)

The most liked stuff for my own use ishexagon/oval/eccentric but a bit too much hard to understand for now.

alkabal, 2020-05-12:
[orientv2.comp.txt](https://github.com/LinuxCNC/linuxcnc/files/4612692/orientv2.comp.txt)

andypugh, 2020-07-13:
I have been keeping this in my inbox hoping to get round to looking at it. 
I still intend to, at some point.

alkabal, 2020-07-13:
No worries i'm not active at this moment but i plan to come back later and i check my mail regularly

I hope this code can be usefull, from my test this work correctly with my spindle using kbmg-212d 

I'm sure can do something really really better !

imo the advantage of this attempt are : 
jog mode from wheel/pendant
"easy" copy past config with minor edit needed
include a possible solution for spindle homing with machine homing or with first M19

Br

alkabal, 2020-11-16:
Hi

Well i now try to review my code.

One things is not managed if you use reverse direction with a relay or two different input on a VFD this is not managed, related to this the vel_speed sent to PID need in this situation to be always positive. (first attempt done with a +-10V analog spindle controller kbmg-212d) 

One clarification is needed for mode 0 shortest move, in this mode the orientv2 process turn the spindle cw and ccw several time until stopping in the tolerance (0.5° is the shortest tolerance for my 400ppr encoder 0.9° per pulse) **i need to know opinion from real user about that.**

mode 1/2 CW and CCW need other approach, imo using actual velocity vs something like delay/velocity needed for the spindle to stop
Unfortunately i have no real idea how to do this for the moment, (i hope you can suggest something easy)

Br