LinuxCNC/linuxcnc #726: Plasmac - 4mm Hole Extension Constant

KimPrzy, 2020-04-13:
Moving 4mm hole extension constant from source to user accessible material cut parameter would be helpful.

Some slower (especially XPR170  via RS-422) or older (dumb/relays) supplies require more time than others to extinguish arc and can drop Arc Ok prematurely. End of motion with arc attached to material results in hole defect (often while arc is mostly attached to bottom of material) and immediate Z+ motion can "drag" it towards/to top of material.

Increasing existing 4mm constant in source produces cleaner holes when increase is (at least initially, for me) proportional to amps.

Guess: preforming extension at feedrate (instead of slowdown) would distribute remaining arc energy over a larger area and might result in lesser defect where arc finally detaches from the material.

phillc54, 2020-04-13:
Is this in reference to #<holes> = 2 in a gcode file.
Is it possible that Pause At End in the Run Panel is causing this issue.

phillc54, 2020-04-14:
I have just done some testing and it is easy to add another parameter to the gcode file so something like:
#<holes> = 2
#<oclength> = 6.5
would then give an extension of 6.5mm. If #<oclength> was not specified then it would default to 4mm. Would this satisfy your requirements.

KimPrzy, 2020-04-14:
Thank you, the addition will work well using value calculated in post. Optional trailing "%" (diameter) would result in less/cleaner gcode for parts with many holes (and support dumb post processors).

Haven't used Pause at End.

Extensions inversely proportional to material thickness and density appear to produce least hole defects. Being submerged or flooded appears to have little positive effect except aluminum. Pierce, lead, hole, torch off, and long/fast extension always produce best small holes in thin material with two different power supplies/torches. 

Worst holes/parts here are now better than local commercial table with bad controller (mach 3) and AHC.

phillc54, 2020-04-14:
Sorry, I am not sure what you mean by:
"Optional trailing "%" (diameter) would result in less/cleaner gcode for parts with many holes (and support dumb post processors)."

KimPrzy, 2020-04-15:
"oclength" you've proposed is great and I'll use it. Thanks!

phillc54, 2020-04-15:
Change has been pushed.

closed 2020-04-15 by phillc54