LinuxCNC/linuxcnc #728: Not understandable behaviour of the Ocode interpreter

R0m3oF0x, 2020-04-14:
## Here are the steps I follow to reproduce the issue:

1. create in your macro directory the folder sub_progs
2.  copy attached files in the folder sub_progs
3.  reroute M6 to toolchange.ngc
4. pulse digital-in-02 low-wait-high
5. while waiting in digital-in-2 on low level do a M6 
 
## This is what I expected to happen:
Because the lines o317 are commented out, the program should continue running after 11 seconds at the latest, regardless of the level at the digital-in-02. Or earlier, if digital-in-02 goes high.
Then the subroutine o320 is executed if #<_current_tool> EQ #<_selected_tool>

## This is what happened instead:
The o320 subprogram will not start anymore. 
The subprogram will only be executed if digital-in-02 is already high at the time of the call.

## It worked properly before this:
Adding the Line: M66 P2 L3 Q11			    

## Information about my hardware and software:
 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian 10 Buster
 * I am using this kernel version (shown by `uname -a`): Linux cnc 4.19.0-6-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.67-2+deb10u2
 * I am running ...
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.9.0-pre0-1350-gb0a0416d3
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): gmoccapy
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
Mesa 7i76ed

=======================================================
Apart from the behaviour described above, there is another behaviour in line 96 "G53 G1 Y100.0 F#29" which I cannot understand. If the "G4 P0.2" is removed from line 99, my milling machine stops in the middle of the movement from line 94 "G53 G1 Z-40.0 F#28". 
As an alternative to "G4 P0.2", line ;(DEBUG, Tool #<_current_tool> -> #<_selected_tool>) can be activated to allow the program to run correctly.	

Please ask if something is still unclear, I will answer as soon as possible.

andypugh, 2020-04-14:
I am not seeing any attached files. You might have to insert the code in to the report.

R0m3oF0x, 2020-04-14:
Thank you very much for the hint, I hope that it worked out now.