LinuxCNC/linuxcnc #752: Plasmac - M190 Numeric Parameter With Leading Zero(s) & New Material Creation

KimPrzy, 2020-04-22:
Leading zero(s) can cause a valid M190 numeric parameter to not match a stored plasmac material number.

Would normally work around in post, but no docs, deaf tech support, and the leading zero(s) are legal and work elsewhere.

M190 P001 - match
M190 P002 - match
M190 P003 - match
M190 P009 - fails 
M190 P015 - fails
M190 P036 - fails

Editing NC to remove leading zero(s) produces correct matches.

Unrelated: Opened material file in editor while chasing match issue. Some newly created materials have negative pierce delay value that displays in app as 0.0 until the entry is changed by user (after which the correct value appears in app/file). Didn't try to recreate or cut when the bad values were present. All were different (but very large) negative values.

phillc54, 2020-04-22:
"Some newly created materials have negative pierce delay value"
I saw a similar problem and it was caused by the same strange value in <MACHINE>_run.cfg file. Could you check if you have the same.

KimPrzy, 2020-04-22:
No error messages. It appears to be a screen update issue and not a material match failure.

Opening file with M190 P001, P002, P36 (and others) causes immediate on-screen update of material displayed in run panel to that which matches the passed parameter. Parameter P036 [and others with leading zero(s)] do not trigger screen update - leading one to believe there was no match and/or that the passed material is not correctly selected. The displayed material will latter change to match the passed material parameter when the loaded NC file is executed.

Was a large nest so P036 was edited to P36 and file reloaded to verify that the correct material was being selected before starting cut. Above findings were in Axis simulator on another machine.

Add new material: Nothing abnormal in the _run.cfg file.

Populating new material entry from currently displayed/existing material instead of default would save time when there is already a very similar material.

Config panel -> PID P Gain (Speed) - On-screen "THC Reaction Speed" tip references plasmac modes 1, 2, and 3 instead of 0, 1, and 2.

andypugh, 2020-04-22:
Just a thought: A leading 0 in a number is interpreted as Octal by Python. 

Python 2.7.16 (default, Apr  6 2019, 01:42:57) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> print(036)
30

Could it be that?

phillc54, 2020-04-23:
I have tried many times but cannot get it to fail. Could you post a copy of your material file on pastebin or maybe the forum so I can give that a try.

Currently a new material uses the parameters of the Default material but I like your idea, I will have a look.

1, 2, and 3 instead of 0, 1, and 2... Thanks, I will fix that.

KimPrzy, 2020-04-23:
Material and NC [here](https://pastebin.com/eCSepnx5). Thanks!

phillc54, 2020-04-23:
I used your gcode snippet and your material file and still cannot see this issue.
Is it possible it is different python versions in different distributions?
I am testing on a VM running the LinuxCNC Debian Stretch ISO with Python 2.7.13

I have pushed an update with better error handling and new materials from the active one as you suggested. Also the materials now display in ascending order.

KimPrzy, 2020-04-23:
Don't see the update yet but new material from old will be a time saver. Thanks!

Removing leading zero(s) (by any means, including hack that appears below) from the passed M190 parameter eliminates the material match issues I've experienced. Casting string to integer to string and latter to integer must evaluate differently than the string to integer on the following line. I don't know python, but am curious if anyone cares to explain this feature.

plasmac_gcode.py @ apx 223 (in "first pass")

# if invalid material number
material = str(int(material))
if int(material) not in materialList:

Installed from stretch ISO per plasmac-user-guide on both machines and VM with buildbot updates.

user@LINUXCNC-TEST:~$ uname -a
Linux LINUXCNC-TEST 4.9.0-12-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.210-1 (2020-01-20) x86_64 GNU/Linux
user@LINUXCNC-TEST:~$ python -V
Python 2.7.13
user@LINUXCNC-TEST:~$

phillc54, 2020-04-24:
Fixed and pushed:
The leading zeros
Kerf Width - I did 0.0001 steps, is that OK?
X and Y Single Cut steps are now 0.0625

I will look into the Material status, that is another good idea. I can probably do it for Axis but not sure about Gmoccapy.

Thanks for the reports and ideas...

KimPrzy, 2020-04-24:
Tried Gmoccapy sim several times but its not friendly/intuitive (to me).

G02 hole should raise error or warning - other parts/rest of the toolpath may also be junk due to open profile, etc. in CAM or bad post processor. Operator (especially someone who didn't CAM the job and/or who can't read gcode) at least needs to know there is an issue.

`elif (line.startswith('g2') or line.startswith('g3')) and line.replace(' ','')[2].isalpha():
                if holeEnable:
                    check_if_hole()
`

phillc54, 2020-04-24:
Do ypou mean something like this:
[http://home.iprimus.com.au/phillcarter/plasma/hole_warning.png](url)

KimPrzy, 2020-04-24:
Yes, exactly. That one little feature could prevent bad cut into high dollar plate for someone new or not paying attention. I'd word it more towards revisiting CAM to locate root issue than alluding to a G02 to G03 edit - just to head them in the right direction. Quick G02/G03/arc swirl/broken profiles blurb in Hole And Small Shape Cutting might help new user understand how cuts end up turned inside out and the resulting warning/error. 

No linuxcnc updates here since prior to your material fix for some reason. Going to try installing in a new VM.

```
Hit:1 http://security.debian.org/debian-security stretch/updates InRelease
Ign:2 http://ftp.debian.org/debian stretch InRelease                           
Hit:4 http://ftp.debian.org/debian stretch Release                             
Hit:5 http://buildbot.linuxcnc.org stretch InRelease                           
Ign:3 http://mirrors.edge.kernel.org/debian stretch InRelease                  
Hit:7 http://mirrors.edge.kernel.org/debian stretch-updates InRelease     
Hit:8 http://mirrors.edge.kernel.org/debian stretch Release                    
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.

```

andypugh, 2020-04-24:
Closing this. You can continue to discuss in here, but it won't show as an outstanding issue any more.

closed 2020-04-24 by andypugh

phillc54, 2020-04-24:
Would you like this for all complete circles or just the ones that qualify as a small hole?

I think the default is 32mm (1.25")

KimPrzy, 2020-04-25:
G02 should probably be warning, not error, for anything that might not be a legitimate G02 outside profile. Side effect is that it could detect small unintended holes not bounded by an outside profile. Error doesn't arise here due to CAM but would be useful to others.

Can't add a new/first material when no other materials saved.

Displayed error when passed M190 material doesn't exist is not clear to operator.

Maybe: Comment on M190 Pn line could optionally pass CAM material settings to plasmac to include while prompting operator for new material or (optionally) update an existing entry. Would bump plasmac up to yet another level while providing infinite pleasure to those constantly tweaking their CAM settings and post :)

`M190 P666 (a=1, n=MS 0.0747in 30A Shielded Duramax, kw=0.04, ph=0.15, pd=0, ch=0.0625, fr=130, ca=30, cv=90, pe=0)`

Action (a): 0 = ignore, 1= add if new material, 2 = add new/update existing material
P parameter=0: ignore action, populate default 00000 material only (one time test cut, etc.)

phillc54, 2020-04-25:
It can't determine what the user intended so it is one of all G2 moves or just G2 moves below the default small hole size. Either way it is only a warning, processing will still continue

"Can't add a new/first material when no other materials saved."
I pushed a fix yesterday that addresses this.

"Displayed error when passed M190 material doesn't exist is not clear to operator."
Yes, I wasn't calling a popup directly but your G2 idea required it so I will move other errors to using popups.

I like your M190 idea, I will look into that when my todo list is a bit shorter

KimPrzy, 2020-04-26:
I couldn't decide on G2, have to think about it some more, and holes in general. Only CAD apps I use (SW and SC) export round holes as two 180 deg. arcs segments - at least to the solid and 2D formats that either CAM will import.

Material additions/update from CAM/code would be awesome. Effort required for set up and cut using plasmac is minimal with material support on controller, like $$$ laser, plasma, and waterjets at work. Thank you for all of your hard work and assistance!

phillc54, 2020-04-26:
I will leave the G2 fot the time being the, in the meantime I will move the other warnings and then have a crack at the materials later...

KimPrzy, 2020-04-26:
Axis:  

Files->Recent Files->Empty Recent File List
or
Files->Empty Recent File List

Everyone would be likely to use it at some point.

phillc54, 2020-04-27:
Axis:
Files->Recent Files->Empty Recent File List

I have created a PR for this.

KimPrzy, 2020-04-29:
Thanks!

I don't know if this would apply to you or anyone else... I set base cut feedrate for all CAM materials to 100. CAM rules that vary cut speed result in a value that I apply in post processor to plasmac.cut-feed-rate as a percentage. CAM material settings are now static (but rules can change) and don't require updates to reflect cut-time settings stored in plasmac's materials.

Created utility to export CAM materials to plasmac format (so that fire is out) and was about close the loop with a utility to update CAM from plasmac when I realized above method would save time and effort far into the future.

phillc54, 2020-05-05:
I have just pushed add materials with gcode