LinuxCNC/linuxcnc #775: galnav.py - divide by zero core dumping Designer

joco-nz, 2020-04-24:
https://github.com/LinuxCNC/linuxcnc/blob/6e7da981d2caa42fa831efad3cd4078a13661f06/lib/python/glnav.py#L404

In line 404 somehow the h value can end up as zero. This results in a divide by zero error and a resulting core dump of designer.
Placing an if statement the line above testing for h=0 and if so setting h=1 solved the issue.
This could well be an edge case but until fixed I could not load a modified version of QTDragon screen that had involved a number of hours of time.

c-morley, 2020-04-24:
You don't happen to have a copy of the file that showed the problem do you?

andypugh, 2020-04-24:
Should be fixed by 12d892e9d4f52b81bbfcd86bb0f3f1ed8022ff84

joco-nz, 2020-04-25:
I will see if I have. It was a version of qtdragon.ui I was working on. But subsequent changes may have fixed what was causing it. 

I also had to ensure fov did not go to 0 as that bombed me a bit later since it is used in a calc as a multiplier that if 0 causes another divide by zero in the same function. 

Cheers,
J.

c-morley, 2020-04-25:
That would be great if you could. I have a hunch it's caused by the fake linuxcnc status I added so that the graphics window displays while designer is running. I probably miss set an value.

joco-nz, 2020-04-25:
I’ve gone back over the last 3 versions but it’s not happening anymore.  It looks like some values in the ui file have changed and are not triggering things again.

c-morley, 2020-04-25:
You didn't happen to update since Andy pushed a fix 11 hours ago did you?
Anyway thanks for checking - if it happens again with someone maybe we can fix the root cause too.

closed 2020-04-25 by c-morley