LinuxCNC/linuxcnc #809: python-gtksourceview2 not available in Buster

rene-dev, 2020-04-28:
python-gtksourceview2 is not available on buster, breaking gmoccapy and other gscreen UIs.

dwrobel, 2020-04-28:
>python-gtksourceview2 is not available on buster, breaking gmoccapy and other gscreen UIs.

I'm interested in packaging `linuxcnc` on `Fedora` and starting from `Fedora >=30` we also no longer have [pygtksourceview](https://koji.fedoraproject.org/koji/packageinfo?packageID=4680) available.

andypugh, 2020-04-28:
I think I have the beginnings of a solution:
This works, standalone:

from [gi.repository](http://lazka.github.io/pgi-docs/#GtkSource-4) import GtkSource as gtksourceview

Then the rest of the code (probably) works. This needs the gir1.2-gtksource-N package. 

Now, this works at the command line, standalone. 

```
Python 2.7.16 (default, Apr  6 2019, 01:42:57) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from gi.repository import GtkSource as gtksourceview
__main__:1: PyGIWarning: GtkSource was imported without specifying a version first. Use gi.require_version('GtkSource', '3.0') before import to ensure that the right version gets loaded.
>>> buf = gtksourceview.Buffer()
>>> 
```

But, there is a problem if you try to mix with "Import Gtk" (with a misleading error message)

```
Python 2.7.16 (default, Apr  6 2019, 01:42:57) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.>>> import gtk
>>> from gi.repository import GtkSource as gtksourceview
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/gi/__init__.py", line 39, in <module>
    raise ImportError(_static_binding_error)
ImportError: When using gi.repository you must not import static modules like "gobject". Please change all occurrences of "import gobject" to "from gi.repository import GObject". See: https://bugzilla.gnome.org/show_bug.cgi?id=709183
```

python-gtksourceview2 was on the Sid repository. My intention for 2.8 was to back-port it via our repo.

c-morley, 2020-04-28:
You have to make sure all libraries use gi.repisitory for that to work.

rene-dev, 2020-04-28:
no, gi only provides it for gtk3. look at the warning.
requiering version 2 yields an error.

rene-dev, 2020-04-28:
Im changing all the imports anyway for python3 to gi, its already done here.

rene-dev, 2020-04-28:
which also works for python2 of course.

c-morley, 2020-04-28:
Gi at least used to have python 2 libraries. But you can't mix dynamic (gi) and static (import pygtk)...obviously for gtk3  there are only gi libraries.

rene-dev, 2020-04-28:
it does, but not for gtksourceview

c-morley, 2020-04-28:
Ahh yes. It's a odd ball

andypugh, 2020-05-11:
gtksourceview2 is now available for Buster from the Linuxcnc repository

closed 2020-05-11 by andypugh