LinuxCNC/linuxcnc #812: step_size Axis bug from joint-axis merge

rene-dev, 2020-04-30:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/axis/scripts/axis.py#L3599-L3614

in the joint axis merge, the line
`a = "XYZABCUVW"[a]`
was added.
python3 complains correctly about line 3613
`TypeError: '<' not supported between instances of 'str' and 'int'`
so  before it was cheching for axis X, Y or Z.
the correct fix would be to change the if to
`a in ['X','Y','Z']`
but given that it never worked correctly for the past 5 years(python2 always returns false), and I could not figure out what step_size is actually used for, I wonder if it is still needed.

c-morley, 2020-05-02:
Looks like step_size is used to the slider minimum size? but only if the INI is missing:
DISPLAY", "MIN_LINEAR_VELOCITY"
or:
"DISPLAY", "MIN_VELOCITY"

andypugh, 2020-06-13:
Any reason to not just fix this? 
Proposal:
```
        if f != 0:
            step_size_tmp = min(step_size, 1. / f)
           if a in 'XYZ': step_size = astep_size = step_size_tmp
           elseif a in 'ABC': astep_size = step_size_tmp
           else: step_size = step_size_tmp
```

rene-dev, 2020-06-13:
I proposed a fix, but as it was never executed since the JA merge(even with python2), and I could not figure out what it does, I wasnt sure if it is still needed.

andypugh, 2020-06-22:
Addressed by b3bbb956102606e847a4f4bf8c5f1d9662014c11

closed 2020-06-22 by andypugh