LinuxCNC/linuxcnc #813: Plasmac - Holes

KimPrzy, 2020-05-01:
```
plasmac_gcode.py -> def get_hole_radius(I, J):

radius = math.sqrt((I ** 2) + (J ** 2))
```

Adding kerf/2 (if present in selected plasmac material entry and G41 compensation not in use) to G03 radius provides value closer to intended hole size for latter evaluation against minDiameter. 

Diameter instead of radius in M67 E3 Qn comment might be more convenient for operators at cut time.

```
elif (line.startswith('g2') or line.startswith('g3')) and line.replace(' ','')[2].isalpha():
            if holeEnable:
                check_if_hole()
```

G02 shouldn't be included in plasmac hole processing. CAM will have already applied kerf to **outside** of any profiles it elected to process as a G02, so its definitely not a hole.

phillc54, 2020-05-02:
I agree with the first two items.

Not sure about the last item, I know holes "should" be cut counterclockwise but some new to plasma guys may not know yet plus if someone is using CAM then I think they should use CAM to do all the hole processing rather than relying on PlasmaC. The #<holes> idea was really for folk without CAM or simple CAM like Inkscape.
Maybe a popup warning for G2 holes that fit the small hole criteria.

KimPrzy, 2020-05-04:
Wrote python function (currently a standalone pre-processor) last night that could be integrated into your gcode pass2. Processes closed (with or without line/arc lead) single/multi segment inside profiles. Targets all minDiameter or smaller holes and (optionally) any other (any number/combination of G01/2/3) closed profiles minDiameter * minDiameter or smaller. Also allows (under protest) the backwards G02 move...

Started as means to convert my holes to recognizable single segment arcs but seemed like plasmac could deal with them (and all of the above) directly. Nonrecursive (slightly longer but easier to maintain) version is 150+/- lines and works within current file mode (no tell/seek to rewind) without additional i/o.  

Didn't know python (but most others) so it will take a few evenings to wrap your existing code around it, test, and post a link for your review.

```
IN
N0180 M03 $0 S1 (Start Cut)
N0190 G01X3.149 Y1.2164 (Lead-In-Line)
N0200 G03X3.2234 Y1.1193 I0.1005 J0. (CCW Arc)
N0210 G03X3.149 Y1.2164 I0.0261 J0.0971 (CCW Arc)
N0220 M05 (Stop Cut)

N0250 G00X3.179Y0.867 (Rapid)

N0290 M03 $0 S1 (Start Cut)
N0300 G01X3.179Y0.842 (Lead-In-Line)
N0310 G01X3.2697Y0.842 (Line)
N0320 G01X3.2697Y0.7534 (Line)
N0330 G01X3.3207Y0.7534 (Line)
N0340 G01X3.3207Y0.842 (Line)
N0350 G01X3.4384Y0.842 (Line)
N0360 G01X3.4384Y0.843 (Line)
N0370 G01X3.3722Y0.843 (Line)
N0380 G01X3.3207Y0.843 (Line)
N0390 G01X3.3207Y0.9544 (Line)
N0400 G01X3.2697Y0.9544 (Line)
N0410 G01X3.2697Y0.843 (Line)
N0420 G01X3.1374Y0.843 (Line)
N0430 G01X3.1374Y0.842 (Line)
N0440 G01X3.179Y0.842 (Line)
N0450 M05 (Stop Cut)

```
```
OUT:
n0000 M67 E3 Q60 (hole process)
n0180 m03 $0 s1 (start cut)
n0190 g01x3.149y1.2164 (lead-in-line)
n0200 g03x3.2234y1.1193i0.1005j0. (ccw arc)
n0210 g03x3.149y1.2164i0.0261j0.0971 (ccw arc)
n0220 m05 (stop cut)
n9999 M67 E3 Q0 (end of hole process)

n0250 g00g03x3.179y0.867 (rapid)

n0000 M67 E3 Q60 (hole process)
n0290 m03 $0 s1 (start cut)
n0300 g01x3.179y0.842 (lead-in-line)
n0310 g01x3.2697y0.842 (line)
n0320 g01x3.2697y0.7534 (line)
n0330 g01x3.3207y0.7534 (line)
n0340 g01x3.3207y0.842 (line)
n0350 g01x3.4384y0.842 (line)
n0360 g01x3.4384y0.843 (line)
n0370 g01x3.3722y0.843 (line)
n0380 g01x3.3207y0.843 (line)
n0390 g01x3.3207y0.9544 (line)
n0400 g01x3.2697y0.9544 (line)
n0410 g01x3.2697y0.843 (line)
n0420 g01x3.1374y0.843 (line)
n0430 g01x3.1374y0.842 (line)
n0440 g01x3.179y0.842 (line)
n0450 m05 (stop cut)
n9999 M67 E3 Q0 (end of hole process)
```

phillc54, 2020-05-04:
I will be interested to see it. It is funny how folk like/dislike certain things, I have an obsessive dislike for line numbers in gcode, I don't know why but...
I also don't know why I am apprehensive about the G2, I figure that as soon as I make it an error someone will want to cut a heap of 20mm discs...
I am all but finished on the gcode material adding/editing.

KimPrzy, 2020-05-04:
Have several more ideas that I'll code as (at least) proof of concept or suggest after more thought.

With two best plasma CAMs and one bad/popular one I see benefits in moving some of their basic functionality to a controller that is available to everyone.

"SH" fork - plasmac that cuts *everything* backwards - with claim that arc rotates in opposite direction in southern hemisphere.

Big metal shop near Boston, MA runs everything through a demagnetizer before cutting, saying there is less bevel in thicker plates. I think its true.

phillc54, 2020-05-05:
I have pushed fixes for the first comments.

closed 2020-05-05 by phillc54

KimPrzy, 2020-05-05:
Nearly done [here](https://pastebin.com/06J0pyxS). Strayed a little (actually quite a bit) from original intent. Try in IDE with pierceonly = true or false. Only tested in IDE (visual studio code) with included gcode. Ignored metric/inch up until now. After material match is selected is the plasmac material name accessible to append as m190 comment? For closed profiles I'll have overcut ignore lead and coast oclength along first segment profile, as option in CAMs do the same thing. Doesn't modify original gcode except to shift Z to line's existing/new comment, add comments, and restore missing axis to modal moves (makes program shorter). Hooks in place to do anything else I can think of. Want to add dynamic pierce for thick material (or making videos). Could add non-default G02 fix to shift kerf (based on selected material) inside and cut as G03 - since there is a reason for their appearance, if that would help the CAM impaired. Queue in python library broken so LNct workaround for now.  Will add many more comments. My second python program (both in the past week), books on order, so go easy (but feel free to pick it apart).

phillc54, 2020-05-06:
I won't get to fully test for a few days but I like the way you are processing it in one pass. To add the new material add/edit we may need to ensure any material magic comments are before any gcode and ignore later ones, that way all file and widget updates are done before gcode processing.
You are setting the feedrate twice, once with a multiplier and again with M67 E3 Qnn
The only issue I have is that it requires the queue module which is not a default in Python 2.7 (where it is named Queue) in the Stretch ISO. This stuff is hard to explain to new users...

KimPrzy, 2020-05-06:
[Here](https://pastebin.com/XMPVFUXj). No longer readable but will be next time you see it (and no massive functions...). Contains extra lines for tests and other stuff that will go away. Junk gcode with last post - just needed profiles larger and smaller than minDiamater quickly. Don't want to make a second pass, obtaining material name was to add to M190 line comment, which won't add to functionality. I'll start using 2.7 and probably ditch queue as FIFO will only add a few lines and not change or break unexpectedly. First python books will be here tomorrow, after which code should hopefully improve. Easy language, which leads to messy code (direction mine took last night).

phillc54, 2020-05-07:
You could do something like:

import sys, re, math
if sys.version_info[0] < 3:
    import Queue as queue
else:
    import queue

It runs fine on Python2 and Python3 here.

KimPrzy, 2020-05-08:
I'll try that tonight, thanks!

Plasmac user guide
11.10. Cut Types

`"Adding a line <pierce-only> = 1 anywhere..."` Missing pound (#) isn't immediately obvious to a new user.

phillc54, 2020-05-08:
Thanks, every time I look at it I see something wrong :(

KimPrzy, 2020-05-08:
Guide is clear, understandable, and not dry and scary like most linux related docs. I'm excited to try the new material scraping. It should make life easier for people as posts are created to support it. Temporary use of default was really good idea and will probably see the most use as people discover how convenient it can be - but syncing all material with plasmac has greatest potential be he big time saver for those who spend a lot of time cutting.

One major fix (which wasn't yet effecting output), a bunch of little ones, and pierce-only. Happy that queue can be used as code will be shorter. Simplifying as much as possible with bias towards language's built in functionality instead of external libraries subject to change.

[Here.](https://pastebin.com/fVz1fr2E)

KimPrzy, 2020-05-12:
[Here](https://pastebin.com/M8d94U45).
`#<holes> = 1 and 2` should also catch G01/2/3 "holes" < minDiameter (haven't tried those, single segment G02/3, or without lead in yet - probably one or more broke..)
`#<profiles> = 1` (overcut at ocLength for now)
`#<arcs> = 1` (slowdown < arcRadius)
`#<arc-radius> 2.0 etc.`
`#<arc-velocity> = 85 etc` percent for M67 E3 Qn
Faked hole G02/3 overcut by following next (first in profile) segment  for its length until time to finish (its going to change to accommodate ramps).
`#<arc-ramp>` will soon do ramp slowdown with default/optional speed/distance/kerf bump. 
Profiles will be 1=both, 2=inside, 3=outside soon.  Above worked here with holes, profiles, and arcs active (only tried on one file). Little time (students all week). Cleaning up/breaking up largest function so rules will be separate sections and everything in manageable screen size chunks (and smoother logic). Easy to add and maintain rules having access to any part of the cut, add something to queue before/after line, cut, or M03/M05, etc. Added queue for warnings, etc. that will output as comments before code. Ramping speed across single/multiple segments and look ahead was reason for queues initially. `VERBOSE=False` will make output quiet. Finally got update with "clear recent files", will be less confusing, especially early in AM or more than one operator! Maybe something in config or individual materials to keep current record(s) from being overwritten by material magic, or setting to always direct current job to default material. Latter would be awesome for new material/nozzle/amps in that it could be tweaked there and then added as a new material(s) with different (than specified in gcode) material number(s).

KimPrzy, 2020-05-15:
[Here](https://pastebin.com/t8pFW1QG). New, shorter, twisted logic gone, no external queue, no huge functions, simple to add/change rules or extend functionality. Guessing on inside/outside profile until code added to check if pierce is in/out of profile and/or cut direction. Haven't checked math yet.

Problems with new/old material magic went away after reinstalling linuxcnc in another VM.

phillc54, 2020-05-20:
Sorry, I haven't had a chance to look at this yet. I will get to it.

KimPrzy, 2020-05-21:
Issue getting one CAM's post to output a comment string longer than 127 characters...

Below processes a material line, regardless of parameter case, sequence, or white space while processing multi line, preserving material name case, loading variables, raising exception for missing mandatory parameters, raising exception for any bad numeric values, etc. 

```
tmpList = []
kw = jh = jd = th = ca = cv = pe = gp = cm = 0.0;
line = '(o=1, NU = 2, Na=5mm Mild Steel 40A, Ph=3.1 , pd=0.1, ch=   0.75,\n fR  = 3000, kw  =  0.5, TH=1, ca=45, cv= 110 , Pe=0.1, Gp=5, cm=1 , JH=0, jd=0)'

# Mandatory variables don't yet exist. After line and from resulting list they and others
# are created, converted to float or int, raising exception if missing or a non-numeric value,
# and finally a check if nu and na are present. Since non-mandatory variables were 
# created above they'll fall through with proper numeric values (subject to newly assigned
# values from line).

if (line.lower().index('o=')):
    try:
        i = re.findall(r'([a-z]{1,2})\s*\=\s*([\S\ ][^,]*)\s*[\,|\)]\s*',line,re.IGNORECASE+re.MULTILINE)
        for a in i: exec (a[0].strip().lower() + "='"  + a[1].strip() + "'")
        ph = float(ph) ; pd = float(pd) ; ch = float(ch) ; fr = float(fr) ; kw = float(kw)
        th = int(th) ; jh = float(jh) ; jd = float(jd) ; ca = float(ca)
        cv = float(cv) ; pe = float(pe) ; gp = float(gp) ; cm = float(cm)
        err1 = nu + na # raise exception w/o material number or name from line
    except:
        print('bad or missing value in magie material line')
        exit()
#latter...
    for i in [na,kw,th,ph,pd,jh,jd,ch,fr,ca,cv,pe,gp,cm]:
        tmpList.append(i)
```