LinuxCNC/linuxcnc #814: Axis DRO background not updating for offsets

rene-dev, 2020-05-01:
Axis DRO has a background, which changes in size depending on the font,
but doesn't take the offsets into account when shown, making them hard to read.

[image]

andypugh, 2020-05-01:
If all the DROs were transparent it would be worse.
If they were all opaque it would be a lot worse 

What would be better? (I don't think now is better) perhaps colours? Cone in blue?

c-morley, 2020-05-02:
For some background info: offsets was disabled in AXIS's library when the DRO page was added.
I re-enabled it later for gremlin/gladevcp. Unfortunately at the time I couldn't figure out how to disable it in AXIS only. The overlay not changing size is probably something I missed to re-enable.
So to summarize the offsets option is really not supposed to be there (in AXIS) as the DRO page gives the same info.
If you want to keep the behavior but fix the overlay, it's probably in  glcannon.py