LinuxCNC/linuxcnc #816: gtk2 not working on python3

rene-dev, 2020-05-03:
currently no gtk2 UI works with python3, as gi for gtk2 is not supported on python3.
they require porting to gtk3.
gremlin requires this gtk3 PR to be merged: https://gitlab.gnome.org/GNOME/gtk/-/merge_requests/1790

starter: https://github.com/LinuxCNC/linuxcnc/tree/gladevcp_gtk3

rene-dev, 2020-05-05:
my wip branch: https://github.com/rene-dev/linuxcnc/tree/rene-gtk3

rene-dev, 2020-05-21:
new wip branch: https://github.com/rene-dev/linuxcnc/tree/limuxcmc

ShadeTechnik, 2020-06-24:
Finally a new attempt to move gladevcp to GTK3. Thank You!

gmoccapy, 2020-09-23:
Gmoccapy is working with gtk3 and Python3 in the named branch.
There are still bugs to resolve, but many work has been done. 
I will continue asap, code is ugly on some parts and not all features are working perfekt. 
Norbert

rene-dev, 2021-05-24:
most parts have been ported to gtk3 now.

closed 2021-05-24 by rene-dev