LinuxCNC/linuxcnc #820: python3 distro support, and dependencies.

rene-dev, 2020-05-03:
which distros can we support on python3?
which dependencies are required(boost, bindings, gtk, tk)?
please test what you are using:
`./configure --with-python=python3 --with-boost-python=boost_python3-py37`
start sim/axis, and run the tests.
so far I only tested on Buster, with python 3.7.3
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions

ozzyrob, 2020-05-03:
Tried with Mint-19.2 (ubuntu bionic)
`uname -msr 
Linux 4.19.106-rt44-lcnc x86_64'

`./configure --with-python=python3 --with-boost-python=boost_python3-py36`
 configure runs ok had to install python3-tk from bionic repo and python3-yapps from focal repo.
As the build fails what information would you like before I add a whole of info that is relevant 
The following is a snippet from when the build fails
[failed.txt](https://github.com/LinuxCNC/linuxcnc/files/4569632/failed.txt)
libbost version info
[libboost-versions.txt](https://github.com/LinuxCNC/linuxcnc/files/4569634/libboost-versions.txt)

TurBoss, 2020-05-03:
Hello,
Thanks @rene-dev

  This is what i get on debian 10
./configure --with-python=python3 --with-boost-python=boost_python3-py37 --with-realtime=uspace
 gives also the same result
http://dpaste.com/1EQKJ22

rene-dev, 2020-05-03:
its trying to link python2 boost, run with make V=1 to see the compiler invocation.
run ls -la /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost*
also, make sure to run make clean before building.

ozzyrob, 2020-05-03:
hmmmm may have to try again on the desktop
Ran this on the laptop:
`./configure -with-python=python3 --with-boost-python=boost_python3-py36`
`make clean`
`make --jobs=2`
`sudo make setuid`
`scritpts/runtests`

Runtest: 233 tests run, 229 successful, 4 failed + 0 expected
Failed: 
	/home/rmurphy/packages/emc-2.9/tests/halmodule.0
	/home/rmurphy/packages/emc-2.9/tests/interp/python-self
	/home/rmurphy/packages/emc-2.9/tests/remap/introspect
	/home/rmurphy/packages/emc-2.9/tests/remap/oword-pycall

rene-dev, 2020-05-03:
I know the halmodule fails, can you check what the problem with the other ones is?

ozzyrob, 2020-05-03:
Worked out the issue with the desktop....
Used
./configure -with-python=python3 --with-boost-python=boost_python3-py3.6

Instead of 
./configure -with-python=python3 --with-boost-python=boost_python3-py36

Typical ID-10-Tea error
Tho I got a few more fails
Runtest: 233 tests run, 215 successful, 18 failed + 0 expected
Failed: 
	/home/rmurphy/emc-test/linuxcnc/tests/halmodule.0
	/home/rmurphy/emc-test/linuxcnc/tests/interp/m98m99/12-M99-endless-main-program
	/home/rmurphy/emc-test/linuxcnc/tests/interp/oword-mdi-sub-update
	/home/rmurphy/emc-test/linuxcnc/tests/interp/python-self
	/home/rmurphy/emc-test/linuxcnc/tests/linuxcncrsh-tcp
	/home/rmurphy/emc-test/linuxcnc/tests/linuxcncrsh
	/home/rmurphy/emc-test/linuxcnc/tests/mdi-queue/oword-queue-buster
	/home/rmurphy/emc-test/linuxcnc/tests/mdi-queue/simple-queue-buster
	/home/rmurphy/emc-test/linuxcnc/tests/motion-logger/mountaindew
	/home/rmurphy/emc-test/linuxcnc/tests/motion-logger/startup-gcode-abort
	/home/rmurphy/emc-test/linuxcnc/tests/remap/introspect
	/home/rmurphy/emc-test/linuxcnc/tests/remap/oword-pycall
	/home/rmurphy/emc-test/linuxcnc/tests/remap/remap-io
	/home/rmurphy/emc-test/linuxcnc/tests/t0/nonrandom
	/home/rmurphy/emc-test/linuxcnc/tests/t0/random-with-t0
	/home/rmurphy/emc-test/linuxcnc/tests/t0/random-without-t0
	/home/rmurphy/emc-test/linuxcnc/tests/toolchanger/toolno-pocket-differ/nonrandom
	/home/rmurphy/emc-test/linuxcnc/tests/toolchanger/toolno-pocket-differ/random

ozzyrob, 2020-05-03:
Ok if you bear with me. I'm pretty green when it comes to running the test suite.
What is the way to check the problem with the failed tests ?

andypugh, 2020-05-03:
cd to the /tests/... directory of the failing test. 
re-run it in verbose mode:
`runtests -v .  `             (you can pass any path to the runtests, and it runs all the tests in that path)
If that doesn't make the problem obvious then you might need to look more carefully. 
if you see an "expected" and a "result"
`diff expected result`
Might show the problem. 
Some tests run a Python script that checks the results itself.

andypugh, 2020-05-03:
```
andypugh@holbrook:~/linuxcnc-dev$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.10 (wheezy)
Release:	7.10
Codename:	wheezy
```
Needed to install libusb-1.0-0-dev and Python3 (of course) 

But then 

```
checking for python pango and cairo modules... Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named pango
no
configure: error: Python pango and cairo modules not found!\ninstall with "sudo apt-get install python-gtk2"
```
But: 
```
andypugh@holbrook:~/linuxcnc-dev/src$ sudo apt-get install python-gtk2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-gtk2 is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 233 not upgraded.
```

rene-dev, 2020-05-03:
@andypugh the suggestion by the configure script is incorrect.
`sudo apt install gobject-introspection python3-gi`

andypugh, 2020-05-03:
Looking further, I think this test is wrong: https://github.com/LinuxCNC/linuxcnc/blob/master/src/configure.ac#L1458
The same test is used here:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/configure.ac#L1657
And also fails. 

In both cases deleting the "if" and "else" clause allows configuration to continue. 
The latter requires the install of "python3-tk" so a change of message is probably needed too.

andypugh, 2020-05-03:
Next obstacle: 
```
checking whether the Boost::Python headers are available... no
configure: error: boost::python is required to build LinuxCNC
...
andypugh@holbrook:~/linuxcnc-dev/src$ sudo apt-cache search libboost-python
libboost-python-dev - Boost.Python Library development files (default version)
libboost-python1.49-dev - Boost.Python Library development files
libboost-python1.49.0 - Boost.Python Library

```

rene-dev, 2020-05-03:
its the python version. libboost-python1.49 seems very very old.
look in ls -la /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libboost*
if there is a python3 version.

andypugh, 2020-05-03:
This is Wheezy, It is very old. 

```
-rw-r--r-- 1 root root  94280 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_iostreams.so.1.49.0
lrwxrwxrwx 1 root root     22 Dec 17  2017 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python.a -> libboost_python-py27.a
lrwxrwxrwx 1 root root     22 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-mt-py26.a -> libboost_python-py26.a
lrwxrwxrwx 1 root root     30 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-mt-py26.so -> libboost_python-py26.so.1.49.0
lrwxrwxrwx 1 root root     22 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-mt-py27.a -> libboost_python-py27.a
lrwxrwxrwx 1 root root     30 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-mt-py27.so -> libboost_python-py27.so.1.49.0
lrwxrwxrwx 1 root root     22 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-mt-py32.a -> libboost_python-py32.a
lrwxrwxrwx 1 root root     30 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-mt-py32.so -> libboost_python-py32.so.1.49.0
-rw-r--r-- 1 root root 508958 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py26.a
lrwxrwxrwx 1 root root     30 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py26.so -> libboost_python-py26.so.1.49.0
-rw-r--r-- 1 root root 303608 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py26.so.1.49.0
-rw-r--r-- 1 root root 508958 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py27.a
lrwxrwxrwx 1 root root     30 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py27.so -> libboost_python-py27.so.1.49.0
-rw-r--r-- 1 root root 303608 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py27.so.1.49.0
-rw-r--r-- 1 root root 504564 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py32.a
lrwxrwxrwx 1 root root     30 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py32.so -> libboost_python-py32.so.1.49.0
-rw-r--r-- 1 root root 301624 Feb  4  2013 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python-py32.so.1.49.0
lrwxrwxrwx 1 root root     23 Dec 17  2017 /usr/lib/gcc/i486-linux-gnu/4.7.2/../../../libboost_python.so -> libboost_python-py27.so
andypugh@holbrook:~/linuxcnc-dev/src$ 
```

I tried --with-libboost-python32 but the result was the same.

rene-dev, 2020-05-03:
You do need to run make clean in between builds

dwrobel, 2020-05-05:
>which distros can we support on python3?

I managed to compile and package _(as rpm)_ `linuxcnc` _(excluding qtvcp)_ for all currently supported versions of Fedora in [copr](https://copr.fedorainfracloud.org/coprs/dwrobel/linuxcnc/).

It compiles fine _(except one strange linking issue on  [fedora-32-aarch64](https://copr.fedorainfracloud.org/coprs/dwrobel/linuxcnc/build/1368102/))_ using combination of `gcc 9.3.1` and `python 3.7.7` to `gcc 10.0.1` and `python 3.8.3`.

There is something wrong with generating documentation _(I'm observing a lot of 'Image '' not found' messages)_ - but I'll investigate it later.

 It's still on a very early stage but, in general, looks quite promising.

>start sim/axis, and run the tests.

`axis` works and all tests passes. All of the required changes were submitted as PRs.

satiowadahc, 2020-08-24:
(First time actually running a RIP, so bear with me)
Fresh Ubuntu 18.04 in a virtual machine (Going to attempt to replicate in a little bit on 20.04 on a real computer)

git clone http://github.com/linuxcnc/linuxcnc
sudo apt install pkg-config libudev-dev libmodbus-dev libusb-1.0-0-dev libglib2.0-dev \
pip3 intltool bwidget libtk-img tclx libxinerama-dev libreadline-gplv2-dev python3-opengl \
libglu1-mesa-dev libxmu-headers libxmu-dev

sudo pip3 install yapps

Download the following links
https://packages.ubuntu.com/focal/libboost1.71-dev
https://packages.ubuntu.com/focal/libboost-python1.71-dev
> And all dependencies from focal
> sudo apt install ./* (* Being the above)

cd linuxcnc/src
./configure --with-python=python3 --disable-gtk
make clean
make -j6
../scripts/runtests

```shell
Running test: /home/chad/git/linuxcnc/tests/interp/python-self
--- /home/chad/git/linuxcnc/tests/interp/python-self/expected	2020-08-21 15:55:32.162767303 -0600
+++ /home/chad/git/linuxcnc/tests/interp/python-self/result	2020-08-24 11:44:40.067104363 -0600
@@ -5,8 +5,6 @@
     4 N..... SET_XY_ROTATION(0.0000)
     5 N..... SET_FEED_REFERENCE(CANON_XYZ)
     6 N..... ON_RESET()
-param1 was retained, value =  25.0
-this is self: True
     7 N..... SET_G5X_OFFSET(1, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000)
     8 N..... SET_XY_ROTATION(0.0000)
     9 N..... SET_FEED_MODE(0, 0)
*** /home/chad/git/linuxcnc/tests/interp/python-self: XFAIL: result differed from expected
```

```shell
Running test: /home/chad/git/linuxcnc/tests/remap/introspect
--- /home/chad/git/linuxcnc/tests/remap/introspect/expected	2020-08-21 15:55:32.166767287 -0600
+++ /home/chad/git/linuxcnc/tests/remap/introspect/result	2020-08-24 11:50:45.054206100 -0600
@@ -13,25 +13,6 @@
    12 N..... SET_FEED_RATE(200.0000)
    13 N..... SET_SPINDLE_SPEED(0, 3000.0000)
    14 N..... START_SPINDLE_CLOCKWISE(0)
-debug:  1073741824 logging: 0
-call_level= 1
-argc= 4
-args=( 1.00000 2.00000 3.00000 3.14159 )
-param # 0 = 1.0
-param # 1 = 2.0
-param # 2 = 3.0
-param # 3 = 3.14159
-current oword subname= introspect
-m_modes[0]= -1
-g_modes[0]= -1
-current tool= 3.0 3.0
-feed= 200.0
-speed= 3000.0
-global parameter set in test.ngc: 47.11
-parameter set via test.ini: 3.14159
-locals:  ['a_new_local']
-globals:  ['_a', '_a_global_set_in_test_dot_ngc', '_a_new_global', '_absolute', '_adaptive_feed', '_b', '_c', '_call_level', '_ccomp', '_coord_system', '_current_pocket', '_current_tool', '_feed', '_feed_hold', '_feed_override', '_flood', '_ijk_absolute_mode', '_imperial', '_incremental', '_ini[example]variable', '_inverse_time', '_lathe_diameter_mode', '_lathe_radius_mode', '_line', '_metric', '_mist', '_motion_mode', '_plane', '_remap_level', '_retract_old_z', '_retract_r_plane', '_rpm', '_selected_pocket', '_selected_tool', '_speed_override', '_spindle_css_mode', '_spindle_cw', '_spindle_on', '_spindle_rpm_mode', '_task', '_tool_offset', '_u', '_units_per_minute', '_units_per_rev', '_v', '_value', '_value_returned', '_vmajor', '_vminor', '_w', '_x', '_y', '_z', 'foo']
-params():  ['a_new_local', '_a', '_a_global_set_in_test_dot_ngc', '_a_new_global', '_absolute', '_adaptive_feed', '_b', '_c', '_call_level', '_ccomp', '_coord_system', '_current_pocket', '_current_tool', '_feed', '_feed_hold', '_feed_override', '_flood', '_ijk_absolute_mode', '_imperial', '_incremental', '_ini[example]variable', '_inverse_time', '_lathe_diameter_mode', '_lathe_radius_mode', '_line', '_metric', '_mist', '_motion_mode', '_plane', '_remap_level', '_retract_old_z', '_retract_r_plane', '_rpm', '_selected_pocket', '_selected_tool', '_speed_override', '_spindle_css_mode', '_spindle_cw', '_spindle_on', '_spindle_rpm_mode', '_task', '_tool_offset', '_u', '_units_per_minute', '_units_per_rev', '_v', '_value', '_value_returned', '_vmajor', '_vminor', '_w', '_x', '_y', '_z', 'foo']
    15 N..... MESSAGE(" after introspect: return value=2.718280 call_level= 0.000000")
    16 N..... SET_G5X_OFFSET(1, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000)
    17 N..... SET_XY_ROTATION(0.0000)
*** /home/chad/git/linuxcnc/tests/remap/introspect: XFAIL: result differed from expected
Running test: /home/chad/git/linuxcnc/tests/remap/m30-interaction
Running test: /home/chad/git/linuxcnc/tests/remap/nested-remaps-oword
Running test: /home/chad/git/linuxcnc/tests/remap/oword-pycall
--- /home/chad/git/linuxcnc/tests/remap/oword-pycall/expected	2020-08-21 15:55:32.166767287 -0600
+++ /home/chad/git/linuxcnc/tests/remap/oword-pycall/result	2020-08-24 11:50:45.134206739 -0600
@@ -5,8 +5,6 @@
     4 N..... SET_XY_ROTATION(0.0000)
     5 N..... SET_FEED_REFERENCE(CANON_XYZ)
     6 N..... ON_RESET()
-multiply: number of arguments= 2
-multiply: number of arguments= 3
     7 N..... USE_LENGTH_UNITS(CANON_UNITS_INCHES)
     8 N..... SELECT_TOOL(1)
     9 N..... START_CHANGE()
*** /home/chad/git/linuxcnc/tests/remap/oword-pycall: XFAIL: result differed from expected
```

Runtest: 234 tests run, 230 successful, 4 failed + 0 expected
Failed: 
	/home/chad/git/linuxcnc/tests/build/header-sanity
	/home/chad/git/linuxcnc/tests/interp/python-self
	/home/chad/git/linuxcnc/tests/remap/introspect
	/home/chad/git/linuxcnc/tests/remap/oword-pycall

htasta, 2020-09-07:
Linux Mint 20

Runtest: 234 tests run, 233 successful, 1 failed + 0 expected
Failed: 
	/home/xxxYYYzzz/linuxcnc-dev/tests/build/header-sanity

The issue was me missing the package: libtirpc-dev
After installing that 234/234 were successful

gmoccapy, 2020-09-16:
I just installed MINT 20 on a new SSD and cloned master branch.

I needed to add the following dependencies to configure with python3

autoconf
g++
libudev-dev
libmodbus-dev
libusb-1.0-0-dev
libglib2.0-dev
libgtk2.0-dev
yapps2
intltool
tcl-dev
tk-dev
bwidget
libtk-img
tclx8.4
libreadline-gplv2-dev
python3-tk
libboostt-python1.71.0
libboost-python1.71-dev
libglu1-mesa-dev
libxmu-dev

That enabled me to do configure --with-python=python3

axis runs, other gui not.

Working on gmoccapy will be done ;-)

Norbert

zwieblum, 2021-05-12:
master from git builds fine on devuan/chimaera with python 3.9.2-1 and it looks like its working. Building the html documentation fails.

rene-dev, 2021-05-12:
can you open a new issue about this? just so we can track what is working and what is not.

rene-dev, 2021-06-02:
the dependencies have all been added in the master-gtk3 branch.
the oldest supported distros are  or Debian 10 buster or ubuntu 20.04 / mint 20.

closed 2021-06-02 by rene-dev