LinuxCNC/linuxcnc #823: configure script fails to pick up correct boostpython

rene-dev, 2020-05-03:
see https://github.com/LinuxCNC/linuxcnc/issues/820
`--with-boost-python=boost_python3-py37`
`--with-boost-python=boost_python3-py36`
should configure not be able to figure this out automatically?

rene-dev, 2020-05-03:
on missing dependencies, the configure script suggests the python2 packages:
https://github.com/LinuxCNC/linuxcnc/issues/820#issuecomment-623114002

dwrobel, 2020-05-03:
In Fedora I have to pass: `--with-boost-python=boost_python27`, `--with-boost-python=boost_python37` respectively.

Unfortunately boost-python{2,3}-devel packages does not contain any `pkg-config` `.pc` files which may help to auto-detect it.

```
$ rpm -ql boost-python3-devel-1.69.0-10.fc31.x86_64
/usr/lib64/libboost_numpy37.so
/usr/lib64/libboost_python37.so
/usr/share/licenses/boost-python3-devel
/usr/share/licenses/boost-python3-devel/LICENSE_1_0.txt

$ rpm -ql boost-python2-devel-1.69.0-10.fc31.x86_64
/usr/lib64/libboost_numpy27.so
/usr/lib64/libboost_python27.so
/usr/share/licenses/boost-python2-devel
/usr/share/licenses/boost-python2-devel/LICENSE_1_0.txt
```

However, assuming that in long term `python3` becomes default, that shouldn't be a problem at all.

rene-dev, 2021-06-02:
this is fixed now, but Im not sure exactly which commit fixed it.

closed 2021-06-02 by rene-dev