LinuxCNC/linuxcnc #825: python3: axis: ImportError: 'interpreter' is not a built-in module

dwrobel, 2020-05-03:
Running `axis` on python3 _(3.7.7)_  on `Fedora 31` generates ` ImportError: 'interpreter' is not a built-in module` as follows:
```
$ linuxcnc
LINUXCNC - 2.9.0~pre0
Machine configuration directory is '/home/sw/linuxcnc/configs/my-simulated-mill'
Machine configuration file is 'my-simulated-mill.ini'
Starting LinuxCNC...
Found file(REL): ./my-simulated-mill.hal
Note: Using POSIX non-realtime
Found file(REL): ./custom.hal
note: MAXV     max: 25.000 units/sec 1500.000 units/min
note: LJOG     max: 25.000 units/sec 1500.000 units/min
note: LJOG default: 2.500 units/sec 150.000 units/min
note: jog_order='XYZ'
note: jog_invert=set()
/home/sw/projects/linuxcnc/bin/axis:3768: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
  glnav.use_pango_font(coordinate_font, 0, 128)
PYTHON: exception during 'this' export:
ImportError: 'interpreter' is not a built-in module
```

Interestingly it is not a fatal error as I can mill successfully 'Linuxcnc` test.

rene-dev, 2020-05-03:
I also noticed this message, but don't know where it comes from, or what is affected by it.
some tests use the interpreter module, and they seem to work fine.

rene-dev, 2020-05-04:
according to https://github.com/python/cpython/blob/f680b517e2701e9a3859afb62628a46eccdce17c/Python/import.c#L2153
the problem could be here:
https://github.com/LinuxCNC/linuxcnc/blob/934da1459a0e1ea7ac3d89bb44661fc278280a95/src/emc/pythonplugin/python_plugin.cc#L319

jlama, 2020-07-13:
This happens because Py_IsInitialized == 1 (no idea why) and thus PyImport_ExtendInittab does nothing.

satiowadahc, 2021-01-14:
Writing it down before I loose it, here's the exception: https://github.com/LinuxCNC/linuxcnc/blob/29122049fdbd4452ae07736a84aefac102184f93/src/emc/rs274ngc/rs274ngc_pre.cc#L171

satiowadahc, 2021-01-14:
That's it! This try catch is completely redundant in python3, The module actually gets imported here:
https://github.com/LinuxCNC/linuxcnc/blob/29122049fdbd4452ae07736a84aefac102184f93/src/emc/pythonplugin/python_plugin.cc#L269
So the other one should be able to be removed with a ifdef for python3

jlama, 2021-01-14:
That is not the issue. In Python3 `PyImport_ExtendInittab` doesn't do anything if the interpreter has already been initialized. Then the `gcode` module fails to import the `interpreter` module because it was not registered as built-in.

Instead, when `Py_IsInitialized() == 1`, built-in modules need to be added manually to `PyImport_GetModuleDict()`

satiowadahc, 2021-01-14:
You would be correct, your code worked with no edits needed!

rene-dev, 2021-01-15:
fixed by https://github.com/LinuxCNC/linuxcnc/pull/1042

closed 2021-01-15 by rene-dev