LinuxCNC/linuxcnc #826: python3.8: configure: error: Required header Python.h missing.

dwrobel, 2020-05-04:
I did compile latest version with python3.8 _(default on Fedora rawhide)_ and configure fails as follows:
```
checking whether the Boost::Python headers are available... yes
checking for boost::python shared library... -lboost_python38
checking /usr/include/python3.8/Python.h usability... no
checking /usr/include/python3.8/Python.h presence... no
checking for /usr/include/python3.8/Python.h... no
configure: error: Required header Python.h missing.  Install it, or specify --disable-python to skip the parts of LinuxCNC that depend on Python
error: Bad exit status from /var/tmp/rpm-tmp.ALK00P (%prep)
```

More detailed output:
```
configure:9678: gcc -c -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib
In file included from /usr/include/python3.8/pystate.h:129,
                 from /usr/include/python3.8/genobject.h:11,
                 from /usr/include/python3.8/Python.h:121,
                 from conftest.c:88:
/usr/include/python3.8/cpython/pystate.h:9:10: fatal error: cpython/initconfig.h: No such file or directory
    9 | #include "cpython/initconfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
```

Basically in order to use that header one has to provide appropriate `-I` option from the pkg-config().

I added:
```
export CPPFLAGS="$(pkg-config --cflags python3)"
```
but above requires https://github.com/LinuxCNC/linuxcnc/pull/751 which depends on https://github.com/LinuxCNC/linuxcnc/pull/755.

However, too avoid passing manually `CPPFLAGS` to `configure` it would be better to add a detection of the `python` development package using `PKG_CHECK_MODULES()`.

TurBoss, 2020-05-13:
```
export CPPFLAGS="$(pkg-config --cflags python3)"
```
this allowed me to configure on Arch

rene-dev, 2020-05-15:
I have the same problem on Ubuntu 20.04

rene-dev, 2020-09-07:
@dwrobel can you check if this fixes the issue, if not please reopen.

closed 2020-09-07 by rene-dev