LinuxCNC/linuxcnc #840: Stepconf broken in master with python2

c-morley, 2020-05-08:
build_ini.py has on line 38:
importlib.reload(sys)

Which seems is python3 only

In stepconf.py line 53 there is:
reload(sys)

Would that be python2 only?

changing the build-ini.py to reload(sys) fixes things in python2

I'm actually not not sure of the python version policy for master are we supporting both or transitioning to python3 only?

rene-dev, 2020-05-08:
Oh, looks like I missed that bit. I will take care of it.

rene-dev, 2020-05-08:
I think the idea is to drop 2 once everything works, but Im not sure.

rene-dev, 2020-05-08:
Should be fixed, please test.
I think stepconf should be 2 and 3 compatible, as it is generally used by users, not developers (so stepconf bugs will be fixed in 2.8), and they use 2.8, not master. we should merge the changes into 2.8, so 2.8 and master have the same codebase.

c-morley, 2020-05-08:
Yes it's fixed now -thanks.
probably unrelated but when I compile I see this:

gcc: fatal error: no input files
compilation terminated.
expr: syntax error

but it still compiles fine.

rene-dev, 2020-05-08:
thats probably unrelated. please repoen if there is a problem with stepconf on python3.

closed 2020-05-08 by rene-dev

dwrobel, 2020-05-11:
>Should be fixed, please test.

@rene-dev I found some more issues which should be fixed now by https://github.com/LinuxCNC/linuxcnc/pull/848.

>gcc: fatal error: no input files
>compilation terminated.
>expr: syntax error

@c-morley Could you please share the full output of the following:
 - `$ gcc -dumpversion && echo OK || echo NOK`
 - `$ gcc -dumpfullversion && echo OK || echo NOK`

commit 402f37cc (2020-05-12): Merge pull request #848 from dwrobel/Fix-TypeError-a-bytes-like-objectis-required-not-str-fixes-840

Fix TypeError: a bytes-like objectis required, not'str’ (fixes #840)
commit ad641994 (2020-05-12): Fix TypeError: a bytes-like objectis required, not'str’ (fixes #840)

Fixes the following exceptions:

For python3:
Traceback (most recent call last):
  File "/home/sw/projects/machinekit/linuxcnc/lib/python/stepconf/pages.py", line 91, in on_button_fwd_clicked
    self['%s_finish'%name]()
  File " fhome/sw/projects/machinekit/linuxcnc/lib/python/stepconf/pages.py", line 881, in finished_finish
    self.a.buid_config()
  File "/home/sw/projects/machinekit/linuxcnc/bin/stepconf”, line 795, in buid_config
    self.d.save(base)
  File "/home/sw/projects/machinekit/linuxcnc/bin/stepcont", line 680 [... cut]