LinuxCNC/linuxcnc #855: pyvcp does not start

selroc, 2020-05-17:
## Here are the steps I follow to reproduce the issue:

 1. set DISPLAY = pyvcp in .ini file
 2. start LinuxCNC

## This is what I expected to happen:
Start pyvcp

## This is what happened instead:
option -i not recognized

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian Stretch

 * I am using this kernel version (shown by `uname -a`): Linux CNCMK1 4.9.0-8-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.110-3+deb9u6 (2018-10-08) x86_64 GNU/Linux

 * I am running ...
   * [X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)

 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): linuxcnc-usape 2.7.14

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): pyvcp

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 5i25

c-morley, 2020-05-17:
pyvcp was never intended to be a main display for linuxcnc.
linuxcnc adds -ini to the system arguments (which seems weird i would have thought --ini would have been standard) and the path to the INI.

Anyways pyvcp doesn't expect this and so fails.
You could write a program to strip the system arguments off or use one of the widget libraries made for main screen display.

selroc, 2020-05-17:
I have tried with 'qtpyvcp' but it is not found. Also I have tried pyvcp in a terminal and it does not show the error. The instructions say this:

INI Configuration

QtPyVCP does not require any special INI settings. To set qtpyvcp as the GUI simply edit the INI DISPLAY entry to read:

[DISPLAY]
DISPLAY = qtpyvcp
...

This will show the VCP chooser every time you start LinuxCNC ...

c-morley, 2020-05-17:
qtpyvcp is a separate project that you would need to add to your system.
If you are having problems with qtpyvcp I suggest you ask their project.

in 2.8, linuxcnc supplies gscreen and qtvcp for screen building

selroc, 2020-05-17:
I need a touch screen interface in short time to do my tests, I am resorting to Gmoccapy for this. Sorry.

closed 2020-05-17 by selroc