LinuxCNC/linuxcnc #858: `motion-logger/startup-gcode-abort` fails intermittently

zultron, 2020-05-19:
Looking at the last month of intermittent failed tests, the `motion-logger/startup-gcode-abort` is the most frequent.  #857 documents the second, possibly related, since these both involve occasionally missing lines in `motion-logger` output.

@jepler recently disabled this test in d7710f69.  There's no PR or separate issue for it, but the commit log refers to #49.

The failed test's console log output follows; though the failure is intermittent, it always follows this pattern, with the same two missing lines:
```
--- expected.motion-logger	2020-05-02 18:44:40.561206824 -0600
+++ out.motion-logger	2020-05-02 18:44:43.085198122 -0600
@@ -73,8 +73,6 @@
 DISABLE
 JOINT_UNHOME joint=-2
 FREE
-SET_TERM_COND termCond=2, tolerance=0.000000
-SET_LINE x=-1.000000, y=-2.000000, z=-3.000000, a=0.000000, b=0.000000, c=0.000000, u=0.000000, v=0.000000, w=0.000000, id=2, motion_type=1, vel=4.988877, ini_maxvel=4.988877, acc=1247.219129, turn=-1
 JOINT_ABORT joint=0
 JOINT_ABORT joint=1
 JOINT_ABORT joint=2
*** /home/buildslave/emc2-buildbot/wheezy-amd64/rip-wheezy-amd64/build/tests/motion-logger/startup-gcode-abort: FAIL: checkresult exited with 1
```

zultron, 2020-05-19:
Sample buildbot failures:
http://buildbot.linuxcnc.org/buildbot/builders/1404.rip-wheezy-rtpreempt-amd64/builds/5123
http://buildbot.linuxcnc.org/buildbot/builders/1520.rip-jessie-amd64/builds/3389
http://buildbot.linuxcnc.org/buildbot/builders/1403.rip-wheezy-amd64/builds/4920
http://buildbot.linuxcnc.org/buildbot/builders/1400.rip-wheezy-i386/builds/4919
http://buildbot.linuxcnc.org/buildbot/builders/1520.rip-jessie-amd64/builds/3386
http://buildbot.linuxcnc.org/buildbot/builders/1500.rip-jessie-i386/builds/3384
http://buildbot.linuxcnc.org/buildbot/builders/1520.rip-jessie-amd64/builds/3385
http://buildbot.linuxcnc.org/buildbot/builders/1403.rip-wheezy-amd64/builds/4917
http://buildbot.linuxcnc.org/buildbot/builders/1400.rip-wheezy-i386/builds/4916
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4586
http://buildbot.linuxcnc.org/buildbot/builders/1510.rip-jessie-rtpreempt-i386/builds/3381
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4585
http://buildbot.linuxcnc.org/buildbot/builders/1630.rip-stretch-rtpreempt-amd64/builds/1770
http://buildbot.linuxcnc.org/buildbot/builders/1510.rip-jessie-rtpreempt-i386/builds/3380
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4578
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4576
http://buildbot.linuxcnc.org/buildbot/builders/1510.rip-jessie-rtpreempt-i386/builds/3369
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4573
http://buildbot.linuxcnc.org/buildbot/builders/1630.rip-stretch-rtpreempt-amd64/builds/1756
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4569
http://buildbot.linuxcnc.org/buildbot/builders/1630.rip-stretch-rtpreempt-amd64/builds/1754
http://buildbot.linuxcnc.org/buildbot/builders/1500.rip-jessie-i386/builds/3363
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4564
http://buildbot.linuxcnc.org/buildbot/builders/1630.rip-stretch-rtpreempt-amd64/builds/1749
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4559
http://buildbot.linuxcnc.org/buildbot/builders/1610.rip-stretch-rtpreempt-i386/builds/1744
http://buildbot.linuxcnc.org/buildbot/builders/1630.rip-stretch-rtpreempt-amd64/builds/1744
http://buildbot.linuxcnc.org/buildbot/builders/1530.rip-jessie-rtpreempt-amd64/builds/3356
http://buildbot.linuxcnc.org/buildbot/builders/1610.rip-stretch-rtpreempt-i386/builds/1743
http://buildbot.linuxcnc.org/buildbot/builders/1500.rip-jessie-i386/builds/3347
http://buildbot.linuxcnc.org/buildbot/builders/1520.rip-jessie-amd64/builds/3347
http://buildbot.linuxcnc.org/buildbot/builders/1402.rip-wheezy-rtpreempt-i386/builds/4392
http://buildbot.linuxcnc.org/buildbot/builders/1404.rip-wheezy-rtpreempt-amd64/builds/5076
http://buildbot.linuxcnc.org/buildbot/builders/1520.rip-jessie-amd64/builds/3337
http://buildbot.linuxcnc.org/buildbot/builders/1402.rip-wheezy-rtpreempt-i386/builds/4382
http://buildbot.linuxcnc.org/buildbot/builders/1401.rip-wheezy-rtai-i386/builds/4598
http://buildbot.linuxcnc.org/buildbot/builders/1404.rip-wheezy-rtpreempt-amd64/builds/5133

From this list, it may be noted that the bug affects all `wheezy`, `jessie` and `stretch` builders, regardless of architecture and RT threads.

Curiously, it seems NOT to affect the precise or buster builders.  It would be interesting if someone with visibility into the buildbot could say whether there's something different about those builders, other than the running distro.

zultron, 2020-05-19:
Actually, I somehow missed failures on `buster` the first time around:
http://buildbot.linuxcnc.org/buildbot/builders/1635.rip-buster-rtpreempt-i386/builds/164
http://buildbot.linuxcnc.org/buildbot/builders/1640.rip-buster-rtpreempt-amd64/builds/377
http://buildbot.linuxcnc.org/buildbot/builders/1640.rip-buster-rtpreempt-amd64/builds/365

No `buster-rtpreempt-rpi4` failures in the last month; not sure if that's significant.

zultron, 2020-05-19:
Reminder to self, how to get a long list of builds (had to check out the `buildbot` source code to figure this one out!):
http://buildbot.linuxcnc.org/buildbot/one_line_per_build?numbuilds=5000

zultron, 2020-05-19:
Is there a quick way of telling from a buildbot build which branch a build applies to?

#49 is about aborts and non-deterministic behavior, and there I said the state tags work would fix that.  It looks like state tags has been merged into `master`.

However, it appears (please check my work!) the changes in [1520.rip-jessie-amd64 #3389][1] are based on a recent `master`, but that test still failed in the manner described here.  If that's true, and if it's true that state tags fixes that issue with abort, then this issue isn't related to #49 (again, please check my work!).

[1]: http://buildbot.linuxcnc.org/buildbot/builders/1520.rip-jessie-amd64/builds/3389

SebKuzminsky, 2020-05-19:
The git branch and commit SHA are both Build Properties listed on the build page.

For http://buildbot.linuxcnc.org/buildbot/builders/1520.rip-jessie-amd64/builds/3389 the branch was `master` and the commit was `2a63953a38083bbd2b705f9cdb7f8df042897300`.

zultron, 2020-05-20:
Thanks, @SebKuzminsky.  I'll get my glasses checked soon.  :D