LinuxCNC/linuxcnc #860: HTML Docs master branch do not contain image URLS

rodw-au, 2020-05-19:
## Here are the steps I follow to reproduce the issue:

 1. Go to Master branch docs : http://linuxcnc.org/docs/devel/html/
 2. Open a link that contains images (eg homing configuration)

## This is what I expected to happen:
Images in the document should be displayed

## This is what happened instead:
Images  are not displayed becasue no image URL is included in the HTML source eg.
[image]
This issue affects a number of documents I reviewed so I think it is a global issue

## It worked properly before this:
2.8 branch HTML is correctly formatted and images are displayed

phillc54, 2020-05-29:
That seems like it may be a Buildbot issue, the documents appear to build OK locally.

phillc54, 2020-06-27:
Looks like a missing module on the Buildbot, all images seem to bring up this error.

http://buildbot.linuxcnc.org/buildbot/builders/2000.docs/builds/5554/steps/compile/logs/warnings%20%283630%29

plasmac-linuxcnc, 2020-07-09:
fixed by #901

closed 2020-07-10 by rodw-au