LinuxCNC/linuxcnc #865: "crazy move" when pressing STOP button / ESC

rmu75, 2020-05-22:
1. load a reasonably complex g-code file (torus.png)
 2. run program
 3. interrupt while doing plunge

## This is what I expected to happen:

Motion is aborted

## This is what happened instead:

Feed or rapid move to some pseudo-random location

## It worked properly before this:

unsure.

## Information about my hardware and software:

Tested on RPI4 with linuxcnc 2.9 pre 8756d78d648d77c68952d1737f632d2eaea83cae

rmu75, 2020-05-22:
```
NML_INTERP_LIST(0x30c190)::append(nml_msg_ptr{size=184,type=EMC_TRAJ_LINEAR_MOVE}) : list_size=1001, line_number=2557
emcTaskPlanExecute(0) return 0
emcTaskPlanLine() returned 2558
Issuing EMC_TRAJ_LINEAR_MOVE -- 	 (  +220,+184,    +0,348.289200,239.222333,-59.698367,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,    +1,25.000000,25.000000,750.000000,    +0,    -1,)
Issuing EMC_TASK_ABORT -- 	 (  +503,+12,   +37,)
NML_INTERP_LIST(0x30c190)::clear(): discarding 1001 items
NML_INTERP_LIST(0x30c190)::append(nml_msg_ptr{size=12,type=EMC_TASK_PLAN_SYNCH}) : list_size=1, line_number=2557
emcTaskPlanClose() called at emc/task/emctask.cc:263
NML_INTERP_LIST(0x30c190)::append(nml_msg_ptr{size=184,type=EMC_TRAJ_LINEAR_MOVE}) : list_size=2, line_number=2558
NML_INTERP_LIST(0x30c190)::append(nml_msg_ptr{size=88,type=EMC_TRAJ_SET_TERM_COND}) : list_size=3, line_number=2558
NML_INTERP_LIST(0x85f08)::clear(): discarding 0 items
Motion id 12 took 1.210069 seconds.
Motion id 0 took 0.000001 seconds.
NML_INTERP_LIST(0x30c190)::get(): {size=12, type=EMC_TASK_PLAN_SYNCH}, list_size=2
emcTaskPlanLevel() returned 0
Outgoing motion id is 2557.
Issuing EMC_TASK_PLAN_SYNCH -- 	 (  +516,+12,    +0,)
emcTaskPlanSynch() returned 0
NML_INTERP_LIST(0x30c190)::get(): {size=184, type=EMC_TRAJ_LINEAR_MOVE}, list_size=1
emcTaskPlanLevel() returned 0
Outgoing motion id is 2558.
Issuing EMC_TRAJ_LINEAR_MOVE -- 	 (  +220,+184,    +0,348.289200,263.222333,-79.698367,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,    +2,8.333333,25.000000,750.000000,    +0,    -1,)
NML_INTERP_LIST(0x30c190)::get(): {size=88, type=EMC_TRAJ_SET_TERM_COND}, list_size=0
emcTaskPlanLevel() returned 0
Issuing EMC_TRAJ_SET_TERM_COND -- 	 (  +222,+88,    +0,    +2,0.010000,)
Issuing EMC_TRAJ_SET_TELEOP_ENABLE -- 	 (  +230,+80,   +38,    +1,)
Motion id 2558 took 28.572219 seconds.
Motion id 0 took 0.000002 seconds.
```

rmu75, 2020-05-22:
My suspicion is that when Interp::close is called, _setup.lazy_close is 1, causing the "close" call to be postponed. Changing https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/rs274ngc/rs274ngc_pre.cc#L932 to = 0; seems to avoid that problem, but I'm not sure what lazy_close is supposed to accomplish in the first place.

robEllenberg, 2020-05-22:
A few questions to help pin this down:

   1. Do you have an ON_ABORT_COMMAND in your configuration?
   2. Does it move to a point that it would later hit in the program, or
   someplace it would never go ordinarily?
   3. Does it occur if you set your blend mode to G64 P### Q0.0 (turn off
   NCD)

rmu75, 2020-05-22:
It executes one or two moves that come after those that are/were queued. Will try with G64 P0.01 Q0.0.

rmu75, 2020-05-22:
Another change that makes the problem go away is adding emcTaskPlanReset() after emcTaskPlanClose() in https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/emctask.cc#L260

rmu75, 2020-05-22:
G64 P0.01 Q0.0 doesn't make a difference.

andypugh, 2020-05-22:
Is it this bug again? 
https://github.com/LinuxCNC/linuxcnc/issues/579

rmu75, 2020-05-22:
> Is it this bug again? #579

somehow. reverse-thinking from the fix mentioned in #579, interp::close needs to be called, but in this situation, it isn't, because of lazy_close flag.

rmu75, 2020-05-22:
I can also trigger such a move with G61 exact path mode.

pcw-mesa, 2020-05-28:
Just tested and the bug is not present in 2.7 or 2.8 ( at least for about 20x re-tries )

zultron, 2020-06-01:
Sure, I'll take a look.  Thanks for the heads-up.

andypugh, 2020-06-01:
From the mailing list

I had the chance to take a snapshot of that accident =:O

When I tested about the omitted lines from backend, I run 3D_Chips.ngc from
axis with logging enabled. After having enuf logs collected I hit stop.
First I thought: ok, this time it stops.
... but then it started to run again (stop button of axis was already shown as
pressed) and the tool run across the workpiece.

I uploaded a screenshot: https://www.imagebanana.com/s/1774/QnA1w8ks.html

zultron, 2020-06-08:
See the fix in #589 

It turns out to be important that both @rmu75's original code and @andypugh's `3D_Chips.ngc` use `G64`.

andypugh, 2020-06-08:
To assign credit where due, my message was a copy and paste  from a message from  Reinhard here:
https://sourceforge.net/p/emc/mailman/message/37026304/

zultron, 2020-06-08:
> To assign credit where due, [...]

To do the same, definitely thanks to Reinhard.  Also, it's extremely valuable to have someone with wide familiarity of both code and issues, sifting through the forums to identifying useful information and connect it to issues to be resolved.  In this particular case, multiple issues.  In your case, a long and sustained history.  Thank you.

PR #882 (2020-06-08): Fix #865, '"crazy move" when pressing STOP button / ESC'
This fixes #865.  It includes a test for the problem, and the top commit explains the problem:

```
LCNC [issue #865][1] found a race condition where an abort during
readahead resulted in unintended motion.  The `emcTaskAbort()`
function appears to be responsible for aborting interp, since it runs
`emcTaskPlanClose()`, but that by itself doesn't flush readahead
segments.

When `emcTaskStateRestore()` restored state from the `3D_Chips.ngc`
test program, executing `G64P.1`, the resulting `SET_MOTION_CONTROL_MODE()`
canon call ran `flush_se [... cut]
commit 3d9d9614 (2020-06-08): Merge pull request #882 from zultron/zultron/2020-06-08-lcnc-865-2.8-pr

Fix #865, '"crazy move" when pressing STOP button / ESC'
closed 2020-06-09 by phillc54