LinuxCNC/linuxcnc #876: Need to write all G76 words in lowercase or interpreter fail.

patrice-conil, 2020-05-28:
# The issue tracker is not a support forum
## Here are the steps I follow to reproduce the issue:

 1. Try to use G76 in lathe mode with Gmoccapy
 2. With following gcode
%
(M14X1)
N10 G7
N11 G18
N12 G90
N13 G21

(THREAD1)
N14 M0 (CHANGE TO T5 ON REAR TOOL POST)
N15 T5 M6 G43
N16 G54
N17 G97 S500 M3
N18 G95
N19 G90 G0 X24. Z5.
N20 G0 Z2.5 X13.484  F1.
N21 G76 P1.0 Z-14. I-0.1  J0.1  K1.
N22 X13.27
N23 X13.106
N24 X12.967
N25 X12.845
N26 X12.735
N27 X12.634
N28 X12.539
N29 X12.451
N30 X12.367
N31 X12.287
N32 X12.211
N33 X12.138
N34 X12.068
N35 X12.
N37 G0 X24. Z5.

N38 M5
N39 G53 X0. Z0.
N40 M30
%

## This is what I expected to happen:
An M14x1 thread will be cut 

## This is what happened instead:
Popup error with => P word is missing ...

## It worked properly before this:
It works as expected only with all words in lowercase, .

## Information about my hardware and software:
linuxcnc1:2.8.0~pre1.5914 
SMP PREEMPT RT Debian 4.9.210-1 
Mesa 7i96 with an Omron encoder on spindle.

andypugh, 2020-05-28:
I don't think that the problem is related to case-sensitivity, are you sure that changing the case made the code work? 

I think that you might be misunderstanding G76. It is a multi-pass threading cycle. You do not need the many X moves after the G76, those would be automatic. 

What slightly surprises me is that G76 is modal. But it seems that is also needs a full set of parameters (so being modal isn't useful) 

If you want to make a thread, then the single G76 line is sufficient. 

If you want to program each pass separately, use G33.

patrice-conil, 2020-05-29:
You're right Andy... that's my mistake.
As I'm not a gcode guru, I used LinuxCNC turning post pro in Fusion to generate this sequence ... and I misinterpreted the error message focusing on G76 and loosing line number.
Of course in my multiple tests, I removed the exceeding lines at the same time I lowercased the words.
Sorry for your wasted time on this non error and thank you.

Patrice

closed 2020-05-29 by patrice-conil

andypugh, 2020-05-29:
If that is output from the Fusion LinuxCNC Postprocessor then there is a bug there, as it appears to generate G-code that does not work.