LinuxCNC/linuxcnc #892: Wrong preview of rounded rectangle path in probe-basic.

rmu75, 2020-06-19:
## Here are the steps I follow to reproduce the issue:

* load g-code file
 
## This is what I expected to happen:

display preview

## This is what happened instead:

displayed preview is wrong (see screenshots). actual motion of the machine is ok.

## It worked properly before this:

2.9 from beginning of june (not sure about exact version, to be determined).

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Ubuntu 18.04
 * I am using this kernel version (shown by `uname -a`): 4.19.0-5-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.37-3 (2019-05-15) x86_64 x86_64 x86_64 GNU/Linux
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ × ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.9.0-pre0-2235-g8e7842fa9
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): probe-basis
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): mesa 7i93

Of note: this seems to be related to a G92 X-offset that seems to be as large as the radius of the wrong circle segments. G-Code file to reproduce and experiments with axis and gmoccapy are in the works.

[image]
[image]

rene-dev, 2020-06-19:
can you attach the gcode file?

rmu75, 2020-06-19:
Excerpted g-code showing this issue

[test-circle.txt](https://github.com/LinuxCNC/linuxcnc/files/4804003/test-circle.txt)

jepler, 2020-06-19:
Atfter modifying the file so that AXIS is content to open it (getting rid of use of tool table and fixing modal gcodes at the start of the program), the toolpath may differ from what is intended, but AXIS and task both trace out the same toolpath.

[test-circle.ngc.txt](https://github.com/LinuxCNC/linuxcnc/files/4805721/test-circle.ngc.txt)
[image]

rmu75, 2020-06-30:
Just tested this again with latest versions of qtpyvcp and probe_basic. It seems to be some interaction with G92 offset, if those are canceled (with G92.1), preview looks like it should, setting an offset and reloading the file show wrong preview again.

rmu75, 2020-06-30:
So this is probably not a linuxcnc issue per se, I opened an issue here: https://github.com/kcjengr/qtpyvcp/issues/42

jepler, 2020-06-30:
Feel free to reopen this issue if further investigation indicates a problem in LinuxCNC

closed 2020-06-30 by jepler