LinuxCNC/linuxcnc #910: PcnConf pid_N_maxerror

phillc54, 2020-07-21:
There have been a few issues on the forum that have been fixed by removing pid_N_maxerror from configs created by Pncconf. It seems there is a minor change required.

https://forum.linuxcnc.org/plasmac/39562-help-with-homing-on-dual-axis-plasmac?start=10#174970

c-morley, 2020-07-21:
Does it need to be set properly metric vrs imperial or is it now recommended to not set it at all?

andypugh, 2020-07-21:
I would consider it a parameter that exists to be set as part of the tuning process, and only if necessary.

c-morley, 2020-07-21:
Generally people (particularly new uses) don't tune steppers - Pncconf sets reasonable defaults.
If setting this to zero is reasonable, I can change that. I only set it on advise from Peter - but that was quite some time ago.
I obviously didn't understand it needed to be different for metric vrs imperial.

phillc54, 2020-07-21:
Sorry I can't help Chris, I just noticed this come up a few times so I thought I would report it.

pcw-mesa, 2020-07-23:
I would probably fix it for metric (0.0127 mm or so). Thsi needs to be scaled correctly for metric
 Bounding the PID to 0.0005 mm doesn't give the PID enough headroom to adjust for
timebase difference between LinuxCNC and the stepgen.

The point of this setting is to limit bogus stepgen velocity corrections caused by position 
feedback sample time jitter. That is, say you have 200 usec of jitter on your servo thread and a 
velocity of 10 IPS, a 200 usec late position sample would be off by 0.002", This would 
cause the PID to command a bogus 20% velocity change with the standard PID tuning, 
by bounding the  maximum PID error to 0.0005" we get a 5% change instead.

Note that this is mostly obsolete if the DPLL is used as this reduces the sample time 
jitter to about 250 ns.

c-morley, 2020-07-24:
Ok thanks for explaining Peter. Currently Pncconf doesn't know whether the firmware has DPLL.
I'll add code to adjust the the setting to .0005 for imperial and .0127 for metric

c-morley, 2020-07-24:
Phill - Thanks for bringing it to my attention

c-morley, 2020-07-26:
Hopefully fix be commit 15264fc in 2.8

closed 2020-07-26 by c-morley