LinuxCNC/linuxcnc #911: thcud.comp

daiman444, 2020-07-27:
greetings!
I am using LinuxCNC 2.7.15.
when the component is operating, the cutter jerks up / down when turning on / off the thc, because
if (! enable) {
         z_pos_out = z_pos_in;
         z_fb_out = z_pos_in;

the amount of jerk is equal to:
// Global Variables
variable float offset;

this problem is persistent for the component.

the problem was solved by adding offset to z_pos_in after shutdown. it looks like this:

if (! enable) {
         z_pos_out = z_pos_in + offset;
         z_fb_out = z_pos_in;
thank you for attention!

jethornton, 2020-07-27:
I have no way to test this but it looks ok to me.

JT

petterreinholdtsen, 2023-10-03:
See also #675, another issue related to thcud.

andypugh, 2023-10-03:
It's been a long wait, but I just made this change. Thanks.

closed 2023-10-03 by andypugh