LinuxCNC/linuxcnc #915: 2.9 pre epilog doesn't execute

bobbevins, 2020-08-02:
When running an M6 remap that runs change_epilog in stdglue, prolog and ngc get executed but epilog does not. I put print statements in prolog, they print, they print in the ngc oword but does not print in epilog so it is nopt running the epilog. IT also shows rack_change not callable. It also shows weird star display in preview as shown in image.

## Here are the steps I follow to reproduce the issue:
Run an M6 remap, it runs the preview and you can see the issues in the terminal.

## Information about my hardware and software:
 * I am using this Linux distribution and version Linuxmint 18.3 Silvia
 * I am using this kernel version Linux 4.19.1-rt3 #1 SMP PREEMPT RT
 * I am running a binary version from linuxcnc.org (including buildbot.linuxcnc.org) I built this morning.
 * I am using this LinuxCNC version 2.9 pre master
 * I am using this user interface (GUI) axis
 * I am using this interface hardware vendor Mesa 7i92M-7i76

[image]

bobbevins, 2020-08-02:
See my files: Running a stripped version of rack_change for testing.

[hal file.pdf](https://github.com/LinuxCNC/linuxcnc/files/5012355/hal.file.pdf)
[ini file.pdf](https://github.com/LinuxCNC/linuxcnc/files/5012356/ini.file.pdf)
[remap.pdf](https://github.com/LinuxCNC/linuxcnc/files/5012357/remap.pdf)
[std glue.pdf](https://github.com/LinuxCNC/linuxcnc/files/5012358/std.glue.pdf)
[toplevel.pdf](https://github.com/LinuxCNC/linuxcnc/files/5012359/toplevel.pdf)

bobbevins, 2020-08-04:
Problem was path issue and incomplete subs.

closed 2020-08-04 by bobbevins

andypugh, 2020-08-04:
I like it when an issue solves itself while I am trying to find the time to get round to it :-)