LinuxCNC/linuxcnc #936: change master packages and default to python3

rene-dev, 2020-09-05:
As discussed in https://github.com/LinuxCNC/linuxcnc/issues/819
at some point(after 2.8 release) masters default and buildbot master packages need to switch to python3.
Its not possible for all components of linuxcnc to be python2/3 compatible, especially the UIs.
We need to decide at what time we merge branches into master that break python2 support.

gmoccapy, 2020-09-05:
We should merge the py3 branches directly after release of 2.8And than work all together to solve the remaining problems NorbertVon meinem Huawei-Tablet gesendet-------- Ursprüngliche Nachricht --------Von: Rene Hopf <notifications@github.com>Datum: Sa., 5. Sep. 2020, 11:13An: LinuxCNC/linuxcnc <linuxcnc@noreply.github.com>Cc: Subscribed <subscribed@noreply.github.com>Betreff: [LinuxCNC/linuxcnc] change master packages and default to python3 (#936)
As discussed in #819
at some point(after 2.8 release) masters default and buildbot master packages need to switch to python3.
Its not possible for all components of linuxcnc to be python2/3 compatible, especially the UIs.
We need to decide at what time we merge branches into master that break python2 support.

rene-dev, 2021-08-01:
closed by https://github.com/LinuxCNC/linuxcnc/pulls?q=master-gtk3

closed 2021-08-01 by rene-dev