LinuxCNC/linuxcnc #942: Possibly spurious alert for configuration update.

JetForMe, 2020-09-27:
# The issue tracker is not a support forum

When this [configuration](https://github.com/JetForMe/router-table/tree/spurious-error) has `22.222` as the `[JOINT_4]SCALE` value, LinuxCNC complains about the INI file version needing to be updated. If the value is set to `22`, it launches without complaint.

This seems like a bug.

## Here are the steps I follow to reproduce the issue:

1. Launch linuxcnc with the above-referenced configuration

## This is what I expected to happen:

It should launch.

## This is what happened instead:

It pops up an alert asking me if I want to attempt to auto-update my configuration.

## It worked properly before this:

I'm adding a new axis. Fractional values work for other axes.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.13 (stretch)
Release:	9.13
Codename:	stretch
```

 * I am using this kernel version (shown by `uname -a`):

```
$ uname -a
Linux router 4.9.0-13-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.228-1 (2020-07-05) x86_64 GNU/Linux
```

 * I am running ...
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):

1:2.9.0~pre0.2180.ga14b08a70 

 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Axis

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): Mesa 7i76e

andypugh, 2020-09-27:
I cloned your github and tried the config. 

It worked up to the point that it didn't find your hardware. 

Are you absolutely sure that you are doing the test that you think you are? The script only looks at [EMC]VERSION when deciding whether an update is needed:

https://github.com/LinuxCNC/linuxcnc/blob/master/scripts/linuxcnc.in#L440

JetForMe, 2020-09-28:
I'm not doing any explicit test. All I know is half the time, when I make a change to a value or config parameter, and then re-launch linuxcnc, I get that alert. I'm about to try again, I'll try to post the INI that causes it.

JetForMe, 2020-09-29:
So, now it's allowing it. I checked the behavior a half-dozen times, switching back and forth between `22` and `22.222` and it always failed on the latter. I don't know what's triggering the alert. It probably won't be long before it happens again, and I'll upload that file.