LinuxCNC/linuxcnc #947: edit offsets page broken in gmoccapy using gtk3

rene-dev, 2020-09-30:
offset page loads, but actually changing an offset does not.

gmoccapy, 2020-10-06:
Can be used now, I fixed the cell editing, I still need to do some testing and sure some code clearance, but can be used now with gmoccapy. Some changes will sure brake other GUI, as I changed the color properties from Gdk.color to Gdk.RGBA

The color changes mus be done in all widgets and GUI using colors. This is the new python3 gtk3 color mode and should be used everywhere. Some times it is needed to convert the colors to strings, to store them or passs them to some gtk components, I have introduced in hal_offsetpage  a helper function convert_color to translate RGBA(1.0, 0.0, 0.0, 1.0) to be "#FF0000"

gmoccapy, 2021-05-14:
As gmoccapy will not be developed further I close this issue.
I will code a new GUI in py3 GTK3!

closed 2021-05-14 by gmoccapy