LinuxCNC/linuxcnc #951: hiding mouse pointer not implemented in gtk3

rene-dev, 2020-10-01:
affects touchy and gmoccapy, the code to hide the cursor is commented out.

gmoccapy, 2020-10-05:
I have pushed a fix for gmoccapy to rene-dev/limuxcmc

IMHO it can be implemented also in TOUCHY

gmoccapy, 2020-10-05:
In Touchy it should look like so:
`
        def pointer_hide(self, b):
                if self.radiobutton_mask: return
                self.prefs.putpref('invisible_cursor', 1)
                self.invisible_cursor = 1
                win = self.wTree.get_object("MainWindow").get_window()
                cursor = gdk.Cursor(gdk.CursorType.BLANK_CURSOR)
                win.set_cursor(cursor)

        def pointer_show(self, b):
                if self.radiobutton_mask: return
                self.prefs.putpref('invisible_cursor', 0)
                self.invisible_cursor = 0
                win = self.wTree.get_object("MainWindow").get_window()
                cursor = gdk.Cursor(gdk.CursorType.ARROW)
                win.set_cursor(cursor)
`
Infortunately TOUCHY does not start on my PC, getting a bunch of error messages

rene-dev, 2021-05-24:
fixed in gmoccapy and touchy

closed 2021-05-24 by rene-dev