LinuxCNC/linuxcnc #952: no enter icon in mdi input in gmoccapy in gtk3

rene-dev, 2020-10-01:
the enter icon is missing in the mdi entry.
but clicking where it is supposed to be does work.

gmoccapy, 2020-10-05:
I fixed that, strange that it is not possible to start with

python3 hal_mdi_history.py

results in 
`raceback (most recent call last):
  File ".././lib/python/gladevcp/hal_mdihistory.py", line 28, in <module>
    from hal_actions import _EMC_ActionBase, ensure_mode
  File "/home/gmoccapy/linuxcnc-org/lib/python/gladevcp/hal_actions.py", line 25, in <module>
    from .hal_widgets import _HalWidgetBase

`
As this is not relevant for the widget being embedded,  I vote for closing this issue

gmoccapy, 2020-10-05:
in hal_action.py the line:
from .hal_widgets import _HalWidgetBase
cause the error. deleting the dot fixes that, but it is only a work arround

grandixximo, 2026-10-05:
Fixed in 69e75579c1 (2020-10-05, in 2.9.0): hal_mdihistory now sets the entry icon with set_icon_from_icon_name instead of the removed gtk3 stock icon. Closing.

closed 2026-10-05 by grandixximo