LinuxCNC/linuxcnc #962: Jog not stopping when motion.feed-hold asserted

phillc54, 2020-10-14:
## Here are the steps I follow to reproduce the issue:
While jogging a joint in teleop mode or a axis in coordinated mode then asserting either motion.feed-hold or motion.feed-inhibit.

## This is what I expected to happen:
The jog should stop.

## This is what happened instead:
In coordinated mode the jog continues.
In teleop mode it works as expected.

## Information about my hardware and software:
 * I am using this Linux distribution and version:
phill@stretch  ~  $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.13 (stretch)
Release:	9.13
Codename:	stretch

 * I am using this kernel version:
phill@stretch  ~  $ uname -a
Linux stretch 4.19.0-0.bpo.6-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.67-2+deb10u2~bpo9+1 (2019-11-12) x86_64 GNU/Linux

 * I am running:
A binary I built myself

 * I am using this LinuxCNC version: 
v2.8.0-66-g5c07c7a6f

 * I am using this user interface:
AXIS

 * I am using this interface hardware vendor and chipset:
sim

c-morley, 2020-10-17:
There was some changes with this that i think are wrong.
maotion.feed-inhibit is supposed to stop _all_ motion like as a safety lock.
This behavior was changed by some commits related to your issue.

phillc54, 2020-10-17:
This thread explains that jog is not a feed but having read the motion man page it does state that feed-inhibit relates to jog as well. So it seems that feed-hold and feed-inhibit have different purposes...

c-morley, 2020-10-17:
Yes the pin is badly named. feed-inhib should really be motion-inhbit or motion-interlock.
If should ideally stop all motion, and never be ignored.
IIRC it did wait till a synch move finished and homing was not affected (Which it should)
It's for things like interlocked doors that stop motion.
There is a similar pin spindle-inhibit for the same purpose.
It's important not make the pin less effective.

andypugh, 2020-10-17:
I think that the docs describe the current, but wrong, behaviour.

I see no argument for the machine to be on but for jogging to not work.

c-morley, 2020-10-17:
The docs describe the right behaviour. I added the pin. Interlock is the reason.

andypugh, 2020-10-19:
I would imagine that it is normal to be allowed to jog with the doors open. But I have no experience with industrial machines to be sure.

jethornton, 2020-10-19:
My BP Discovery 308 VMC allows jogging with the doors open, so does my 
CHNC I

JT

c-morley, 2020-10-19:
The door interlock was just an example, because you said you could not think of anything that could required it.
 To say your machine allows jogging with the door open means every machine should, seems to miss the point.
And yes some machine require to be in a special mode to jog while the door is open (I believe HAAS).
I can't think of many industrial machines that don't have interlocks of some kind.
If you don't want to interlock motion with anything then you don't connect the pin to anything.

The behavior of the pin was added specifically to interlock all motion with user connected logic. I never got around to looking into homing but it should really stop that too.
My point is the behavior of the pin has changed, so now it doesn't do what it was intended to do.
If we need to change the name of the pin so it better reflects it's use, then sure let's do that too.

c-morley, 2020-10-19:
I suppose, if your argument is finer grained options, we could add a separate inhibit pin to jogging and homing.
Oh in fact there is homing-inhibit (it's just hard to find in the docs)
So an alternative is to add a jogging inhibit and then a n integrator can decide how much they wish to inhibit.