LinuxCNC/linuxcnc #966: Halcope - Can save data files but not open them

andypugh, 2020-10-21:
This is more an obvious enhancement than a bug. 

Halscope is a really useful tool, and it has the capability of saving data files. (albeit in a rather odd and inefficient format)

It would be very useful (for example when analysing other peoples plots from the mailing list and forum) to be able to load their log files back in to Halscope to zoom in and otherwise analyse. 

I think that there is probably no reason to maintain the current format. Just changing it to CSV (or maybe mf4?) would be an improvement in usability.

rodw-au, 2020-11-21:
I would love to see halscope and halstreamer both use csv files. Maybe there is a command line switch. I think CSV is the more broadly used format. In the past I have analysed data collected this way in Google sheets.

havardAasen, 2021-06-09:
If you are interested, I can submit a PR with a patch that saves the log files in csv format. I also made a python script to create plots, I found it easier then changing halscope.

I found the plot to be quite nice to work with, (interactive html). The only downside is that to run the python script it requires Bokeh library, and you can't 'apt-get' it. For me, the easiest was to install it with pip.

I attached a csv file created with halscope. The html file contains a interactive plot created with the python script. The plot is based on information from the documentation, HAL Tutorial - stepgen example.

GitHub didn't like html files... The dark theme is optional.
[halscope.html.csv](https://github.com/LinuxCNC/linuxcnc/files/6626192/halscope.html.csv)

[halscope_slow.csv](https://github.com/LinuxCNC/linuxcnc/files/6626190/halscope_slow.csv)

andypugh, 2021-06-09:
Saving .CSV would definitely be a start, but what I would really like to see Halscope do would be to import the files that it exports.
Then users on the forum could post their halscope data and we could look at it, and see what they see. 
(CSV would at least allow import in to Excel, but Excel is not great for interactive data plotting)

grandixximo, 2026-01-21:
This could be closed if PR  #3722 is merged