LinuxCNC/linuxcnc #969: obsolete reference to halui.jog in 2.8 documentation

Sigma1912, 2020-10-28:
halui.jog has been removed in version 2.8 but there are still references to it in the 2.8 docs

It's fine there
linuxcnc.org/docs/2.8/html/man/man1/halui.1.html

but it's wrong here
linuxcnc.org/docs/2.8/pdf/LinuxCNC_Documentation.pdf

and here:
linuxcnc.org/docs/2.8/html/gui/halui.html

also here ( I presume this is the doc meant for the current stable release?):
linuxcnc.org/docs/html/gui/halui.html

andypugh, 2020-10-28:
@jethornton Can you answer the question about the link-with-no version?

c-morley, 2020-12-13:
Why did we remove the halui pin reference in the main docs?

dngarrett, 2020-12-13:
Commit 589db06 removed the wrong and the redundant information in halui.txt because it was wrong and reundant to the information in the halui manpage (docs/man/man1/halui.1) which has been maintained much more consistently.   A reference added to halui.txt directs readers to the man page (man halui)

if someone wants to restore halui,txt, they will need to maintain redundant information in two places

c-morley, 2020-12-14:
Thanks for the response.
Why would we remove docs because they were out of date? The usual things is to update them.
The HTML docs are used as reference often , for instance in forums.
Docs that refer you to other docs (especially ones you need to have a machine to see) are not user friendly.

andypugh, 2020-12-14:
The manpages are available as HTML, in this case http://linuxcnc.org/docs/2.8/html/man/man1/halui.1.html

I can see an argument that (in the html docs) a link to the manpage html would be more useful than a command line example. 

That still leaves the case of offline PDF, however.

c-morley, 2020-12-14:
Well having a link the the HTML man pages would have been much better, I agree (don't know there was any)

But I feel that the HTML docs and man pages are two different animals and one shouldn't replace the other.
Man pages are plain, concise at-machine reference.
HTML/PDF docs are more meant to be pretty, expanded versions - but they still should be complete.

I mean an extreme example would be HTML docs that mostly say see the relevant man page -horrible.

I understand the thought process - two sets of docs to stay relevant, but take the halui docs - the reason they were so out of date is because of the HUGE change in linuxcnc as a whole. This is not a normally occurring problem.

personally - If we had offline HTML docs available - i would never use man pages and I already don't use PDF - though if I printed docs I suppose the PDFs would be nicer.

I have replaced the pin references in a branch i have - i just haven't pushed (to master) yet.
I'll wait to see if there is more discussion.

andypugh, 2023-10-03:
Now that we keep a */stable/* fixed URL for the docs this was fairly easy to fix. 
https://github.com/LinuxCNC/linuxcnc/commit/f53a8119449f2cd4854c433537ffcde0fdad109f

closed 2023-10-03 by andypugh