LinuxCNC/linuxcnc #970: Tool Pocket not evaluated correctly in 2.8 stdglue

mydani, 2020-10-28:
## Here are the steps I follow to reproduce the issue:

 1. Open Linuxcnc, ensure tools in tool table are as follows: Tool 1 Pocket 5, Tool 2 Pocket 10
 2. Change tool via M6 to Tool 1, then Tool 2
 3. Use "change_prolog" method to print out the values of self.current_tool, self.selected_tool, self.current_pocket, self.selected_pocket

## This is what I expected to happen:

The "tool" values reflect the tool no. in the tool table
The "pocket" values reflect the pocket no of the corr. tool in the tool table

e.g.
self.current_tool = 1
self.selected_tool = 2
self.current_pocket = 5
self.selected_pocket = 10

## This is what happened instead:

self.current_tool = 1
self.selected_tool = 2
self.current_pocket = 1
self.selected_pocket = 2

It seems the pocket values are the index of the tool table rather than the pocket no.

## It worked properly before this:
- I made it work in a branch prior to 2.8 release, did never work on mainstream (neither 2.7, nor 2.8).
- The correct pocket values are used if the method "self.find_tool_pocket(self.xxx_tool)" is used. This works in sim but unfortunately stalls on a real machine when loading the UI.

## Information about my hardware and software:

 * I am using this Linux distribution: Debian Wheezy
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this user interface: gmoccapy
 * MESA 5i25 + 7i77

mydani, 2020-10-28:
So this is how I fixed it for me - beware, I do not use the standard M6 routine, therefore needed to add some more things. (Also tricky was to find out to use "yield" instead of "return" for all INTERP_... as INTERP_EXECUTE_FINISH will throw a python error (INTERP_EXECUTE_FINISH demands proper python generator).)

import os
import sys
import emccanon 
from interpreter import *
from gscreen import preferences
throw_exceptions = 1

debug = True
if debug:
    pydevdir = '/home/daniel/Dokumente/aptana/Aptana_Studio_3/plugins/org.python.pydev_3.0.0.1484937626/pysrc'

    # the 'emctask' module is present only in the milltask instance, otherwise both the UI and
    # milltask would try to connect to the debug server.

    if os.path.isdir( pydevdir ) and  'emctask' in sys.builtin_module_names:
        sys.path.append( pydevdir )
        sys.path.insert( 0, pydevdir )
        try:
            import pydevd
            emccanon.MESSAGE( "pydevd imported, connecting to Eclipse debug server..." )
            pydevd.settrace()
        except:
            emccanon.MESSAGE( "no pydevd module found" )
            pass

# REMAP=M6  modalgroup=6 prolog=change_prolog ngc=change epilog=change_epilog
# exposed parameters:
#    #<tool_in_spindle>
#    #<selected_tool>
#    #<current_pocket>
#    #<selected_pocket>

def change_prolog(self, **words):
    try:
        # this is relevant only when using iocontrol-v2.
        if self.params[5600] > 0.0:
            if self.params[5601] < 0.0:
                self.set_errormsg("Toolchanger hard fault %d" % (int(self.params[5601])))
                yield INTERP_ERROR
            print "change_prolog: Toolchanger soft fault %d" % int(self.params[5601])

        if self.selected_pocket < 0:
            self.set_errormsg("M6: no tool prepared")
            yield INTERP_ERROR

        if self.cutter_comp_side:
            self.set_errormsg("Cannot change tools with cutter radius compensation on")
            yield INTERP_ERROR

        self.params["tool_in_spindle"] = self.current_tool
        self.params["selected_tool"] = self.selected_tool

        if self.current_tool>=0:
            (status, pocket) = self.find_tool_pocket(self.current_tool)
            if status != INTERP_OK:
                self.set_errormsg("T \"%d\": pocket not found" % (self.current_tool))
                yield status
        else:
            pocket = -1 # this is a T0 - tool unload

        self.params["current_pocket"] = pocket

        if self.selected_tool>=0:
            (status, pocket) = self.find_tool_pocket(self.selected_tool)
            if status != INTERP_OK:
                self.set_errormsg("T%d: pocket not found" % (self.selected_tool))
                yield status
        else:
[... 120 lines cut ...]
        if self.feed_rate == 0.0:
            yield "%s: feed rate must be > 0" % (r.name)
        if self.feed_mode == INVERSE_TIME:
            yield "%s: Cannot use inverse time feed with canned cycles" % (r.name)
        if self.cutter_comp_side:
            yield "%s: Cannot use canned cycles with cutter compensation on" % (r.name)
        yield INTERP_OK
    
    except Exception, e:
        raise
        yield "cycle_prolog failed: %s" % (e)

# make sure the next line has the same motion code, unless overriden by a
# new G code
def cycle_epilog(self,**words):
    try:
        c = self.blocks[self.remap_level]
        self.motion_mode = c.executing_remap.motion_code # retain the current motion mode
        yield INTERP_OK
    except Exception, e:
        yield "cycle_epilog failed: %s" % (e)

# this should be called from TOPLEVEL __init__()
def init_stdglue(self):
    self.sticky_params = dict()

mydani, 2020-10-28:
Pocket behaviour also reported before, https://github.com/LinuxCNC/linuxcnc/issues/400
"Wrong" return instead of yield used reported before as well, https://github.com/LinuxCNC/linuxcnc/issues/217

andypugh, 2020-10-29:
@rene-dev I thought this was fixed?

rene-dev, 2020-10-29:
@andypugh its complicated. the python module just exposes what the c code has, and the c code uses the index internally, like he said:

The correct pocket values are used if the method "self.find_tool_pocket(self.xxx_tool)" is used. This works in sim but unfortunately stalls on a real machine when loading the UI.

so you do need to call this function to get the real number, just like the c code does. why it works only in sim I will investigate.

its more a documentation issue, or issue of the default stdglue.
there should really be a way to use remap without python, as it only makes everything way to complicated.

mydani, 2020-10-29:
BTW. here I did a branch on 2.8 release which provide the pocket rather than the tool table index.
However it's a bigger change....

https://github.com/LinuxCNC/linuxcnc/compare/master...mydani:2.8_pocketfix_on_release

mydani, 2020-10-29:
> @rene-dev 
> why it works only in sim I will investigate

This is not the case anymore, it was a matter of updating the stdglue (yield instead of return, wrong casts, etc.).
With the example code above it is working fine on my physical machine.

mydani, 2020-11-01:
Update from today's debugging:
<_current_tool>, <_selected_tool>, <_current_pocket>, <_selected_pocket> work directly in the change macro without additional magic in stdglue. This renders the call to the find-functions unnecessary.

This works for all tools >0 with pocket values >0. _current_tool is actual tool in spindle, _selected_tool is new tool, _current_pocket is the pocket in the changer the current tool needs to be put back, _selected_pocket is the pocket the new tool needs to be taken from.

Special cases are:
* tool 0 (no tool) => selected_pocket is inconsistent but does not matter
* tools with pocket no. EQ 0  => _selected_pocket will be -1 for these (instead of 0)
* tools with pocket no. LT 0 => _selected_pocket will be as in tool table, _current_pocket will be 0 when currently in spindle

rmu75, 2025-11-23:
@rene-dev: is this done?

rene-dev, 2025-11-23:
I will take a look at this