LinuxCNC/linuxcnc #972: Proposals for further development

Lehmeiers, 2020-10-31:
Hello!
Since I did not know where to go with my suggestions I will try it here:

1. it is a pity that I did not find a possibility to choose between lathe and milling machine. Because the commands (e.g. G82) are partly contradictory.

2. to create a setup sheet, which is generated from the program and is at the beginning of the program file.

3. creating a tool memory so that you can see which tools, with which values, are installed in the machine.

4. the possibility of programming, e.g. directly in PAL, and this, as in a workshop programming with support and dialog guidance.

Here is a simple example :
https://de.wikipedia.org/wiki/Werkstattprogrammierung

5. a control panel with industry-related symbols and their handling.

Since I know that this would be a big change, I want to explain why I came up with these changes.

1. if you could program in PAL LinuxCNC would be an alternative to all the expensive training programs and controls which are used at the moment.
Unfortunately I could not use such a program during my retraining ( which was searched for by many retrainees ) with which I could practice PAL, let alone simulate or even control a whole machine.

2. if you are learning on a controller, then you want to have you on your machine.
So it would be a start for beginners and machine builders.

3. machine manufacturers and machine conditioners could offer an inexpensive control that is up to date.

A machine that is programmed by hand is always more flexible and offers more possibilities than if I have to do it by CAM program. 
There I have to buy one first, then learn it and then load the program into the machine. 
In the hobby area, as well as in the professional area, this would also be a great help.

Maybe you break out in laughter, or ignore it. 
I also know that it won't work in the next version, but it would be nice if it could be implemented in the long run.

You have started a great work - don't give up halfway through.
I am convinced that LinuxCNC has what it takes to beat the other controllers on the market. 
You just have to become a bit more user friendly.
It would be great to experience that.

Translated with www.DeepL.com/Translator (free version)

gmoccapy, 2020-10-31:
I do not understand what is your problem

All you mentioned is possible with LinuxCNC. I do not know PAL, but G Code is very flexible and if you want your own GUI, it is just a few lines of python code. You can look at gmoccapy code as an example. I know a school using gmoccapy for training purpose a Sim config. 

Norbert

andypugh, 2020-11-01:
3) Tooledit? 

4) I am not sure what you mean by PAL, and I can't read German very well. But looking at the pictures, it looks a lot like: 
https://forum.linuxcnc.org/41-guis/26550-lathe-macros
Or:
https://www.youtube.com/watch?v=x7dKkZPYHU0

silopolis, 2020-11-08:
If I'm correct, QtPyVCP also has tool table management and is working on conversational features.

About tool table, I'd heavily support a tool table extension project as well as adding support for tool names in addition to tool numbers. 

About panel, if you happen to have a set of standard icons/drawings I'd be grateful if you could share it 😊