LinuxCNC/linuxcnc #974: Add support for Mesa hardware (7i95 / 7i97 / 7c80) in LinuxCNC 2.8

thomam04, 2020-11-07:
Certain Mesa hardware still requires LinuxCNC 2.9 (development branch) to function. 

Example of hardware: 7i95 / 7i97 / 7c80 

Could we please add support for commonly available Mesa hardware to the hm2 drive in the stable branch 2.8?

andypugh, 2020-11-07:
What is missing from 2.8? I would have expected them to be covered by the hm2_eth and hm2_rpspi drivers.

thomam04, 2020-11-07:
I have tested the 7i95 and it's still not supported in 2.8. 

PCW also mentioned, that the 7i95 / 7i97 and 7C80 still requires 2.9. 

https://forum.linuxcnc.org/27-driver-boards/37894-mesa-7i95?start=120#186919

andypugh, 2020-11-07:
Looking at it, it seems that they would work, except that the driver thinks that the port width is invalid. 
Do you have a way to build LinuxCNC from source or to run the buildbot packages to test? I don't have any of the hardware in question.

thomam04, 2020-11-07:
Yes, I am currently running 2.9 from source due other reasons. I would just need some instructions on what to change/add in the current 2.8 branch.

pcw-mesa, 2020-11-11:
2.8 would need the complete hostmot2 driver set to be updated to  2.9 
to work with the 7C80,7I95,7I97. Its not just the port width, those cards require
support for a new firmware module.

andypugh, 2020-11-11:
@pcw-mesa Do you think we should? 2.8 mainlined Pi support, so it seems a shame not to support the 7C80. The driver is quite well compartmentalised so it can probably be done with low risk.

pcw-mesa, 2020-11-11:
Lots of people are using 2.9 and I have not heard of any driver related issues
so I think its safe. The additions should only affect users with new firmware 
so I don't really see any downside to pulling all of the 2.9 hostmot2 changes into 2.8

andypugh, 2020-11-15:
I have created a test branch, which should end up in the buildbot scratch directory soon. 
http://buildbot.linuxcnc.org/dists/buster/scratch-rtpreempt/binary-amd64/
I have no hardware to test with, so hope that Mesa / EUSurplus / JT can run a few tests. 
https://github.com/LinuxCNC/linuxcnc/pull/978

andypugh, 2020-11-16:
I put in the missing makefile bits later. I don't know if debs got re-made after that.

pcw-mesa, 2020-11-16:
I cannot use the debs so built from the origin/andypugh/2.8_hm2_update branch

andypugh, 2020-11-16:
6b67bdd should have fixed that problem. Can you try again?

pcw-mesa, 2020-11-16:
That problem was fixed with a new fetch and all the new stuff driver seems to be there
One remaining issue is that the hostmot2 manual page does not seem 
to be up to date (for example, it mentions the rcpwm but there is no pin 
description as there is in the 2.9 manual)

andypugh, 2020-11-24:
I can fix the docs. Can you confirm that the pull request works with the hardware?

pcw-mesa, 2020-11-24:
Yes it does work at least with a 7I95 (which probably means all the new stuff is there)

andypugh, 2020-11-29:
f423c5831dd51f32dab485d9ecc11bf2fa02d31c

closed 2020-11-29 by andypugh