LinuxCNC/linuxcnc #982: Spindle using other speed than in GCode, 2.8

mydani, 2020-11-19:
Here's a short clip about the behaviour:
https://www.youtube.com/watch?v=7x6jb6AMuRA

## Here are the steps I follow to reproduce the issue:

 1. Load a GCode with M3 and M4 commands using different spindle speed and place M1 after each command.
 2. Run it once.
 3. Re-Run it - check the speed of the spindle at 1st m3 - it's not what it is set to in GCode.

## This is what I expected to happen:
Spindle always uses the speed given within the GCode.

## This is what happened instead:
Spindle used the RPM last used in the previous program.

## It worked properly before this:
No clue about the version, but it works properly if some axis movements happen before the M3/M4 command.

## Information about my hardware and software:

 * 2.8 vanilla + gmoccapy
 * 3.x RTAI something...
* mesa card

andypugh, 2020-11-22:
I think that this might be a Gmoccapy display problem. 

I ran a test with Axis and got the result I expected (when watching the spindle.0.speed-out pin, as Axis does not have an on-screen speed command display) 

Can you repeat your test with a halmeter on the spindle.0.speed-out pin and report if the problem is seen there too?

andypugh, 2021-01-06:
Any update on this? Can I close the issue?

mydani, 2021-01-09:
Hello Andy,
Sorry for not getting back until now - I am currently rebuilding parts of my work shop. I will test until end of next week!

andypugh, 2025-05-22:
Did you complete rebuilding your workshop? Is this issue still open in 2.9.4 ?

Sigma1912, 2025-05-25:
I just tested on 2.9.4 with the following gcode and cannot reproduce this issue.
```
g0 x0
S100 M3
g0 x5
M1
S200 M4
g0 x10
M1
g0 x5
m2
```

Sigma1912, 2025-05-25:
HOWEVER, testing with this gcode makes strange things happen to the spindle speed:
```
g0 x0
S100 M3
M1
g0 x10
S200 M4
M1
g0 x5
m2
``` 

[image]

This does not happen in AXIS GUI.

This might be fixed with this PR https://github.com/LinuxCNC/linuxcnc/pull/3454.
But I'll have to test this tomorrow.

andypugh, 2025-06-03:
Thanks @Sigma1912