LinuxCNC/linuxcnc #985: NameError: name 'attrgetter' is not defined

satiowadahc, 2020-11-20:
## Here are the steps I follow to reproduce the issue:

 1. Branched off master, compiled with ./autogen.sh && ./configure --with-python=python3
 2. Load any file With G0 and G1 Moves
 3. Double click on the path of Either G0 or G1 Moves

## This is what happened instead:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/qt5_graphics.py", line 902, in mouseReleaseEvent
    self.select_fire()
  File "/usr/lib/python3/dist-packages/qtvcp/widgets/gcode_graphics.py", line 202, in select_fire
    self.select(x, y)
  File "/usr/lib/python3/dist-packages/rs274/glcanon.py", line 540, in select
    min_depth, max_depth, names = min(buffer,key=attrgetter('near'))
NameError: name 'attrgetter' is not defined

## Information about my hardware and software:

 * I am running ... A binary I built myself
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): Custom QTVCP

Thinking I'll be able to look into on Monday if no one else figures it out by then

rene-dev, 2020-11-21:
I cannot reproduce this in axis or qtvcp. which UI did you use?
most likely it is just missing `from operator import attrgetter`

satiowadahc, 2020-11-23:
Can't reproduce today. Maybe I modified something. Closing this for now

closed 2020-11-23 by satiowadahc