LinuxCNC/linuxcnc #991: End of line comments in .ini lead to hanging / errors in 2.8

Hans470, 2020-11-22:
With Linuxcnc 2.7.x it was no problem to have comments in the ini-files and the end of the line like
``` ini
[AXIS_X]
MIN_LIMIT = 0.0 # comment
```
But in version 2.8 it seems to to be supported any longer. The problem is that Linuxcnc then nearly hangs up.

## Here are the steps I follow to reproduce the issue:

 1.
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

andypugh, 2020-11-22:
I seem to remember this as a deliberate change, but I don't recall the reasoning behind it.

phillc54, 2020-11-23:
It is noted in the documents:
http://linuxcnc.org/docs/2.8/html/config/ini-config.html#_comments

Hans470, 2020-11-23:
Ok, but in 2.7 it was still accepted even though the documentation says not.
Maybe this can be checked by the update script to prevent later errors?
It took me too much time to find that the source of the error was only the stupid comment.

rene-dev, 2020-11-23:
Is there any technical reason for the new behavior?

andypugh, 2020-11-23:
Dewey found the commit:

https://github.com/LinuxCNC/linuxcnc/commit/c3ac4f5bf2f76b97

Hans470, 2020-11-24:
Ok but when an ERR_CONVERSION is thrown - why is Linuxnc still starting?

amitSA, 2024-08-08:
[linuxcnc version 2.9.3]

I'm new to linuxcnc, I was adding in-line comments and `linuxcnc my-mill.ini` would hang after printing this:
```
LINUXCNC - 2.9.3
Machine configuration directory is '/home/cnc/Desktop/hal_configs/configs/infinity-mill'
Machine configuration file is 'infinity-mill.ini'
```

I digged deeper and this was an OOM issue.   It would have saved me a couple hours if LinuxCNC printed a helpful error message that my INI file was invalid.

andypugh, 2024-09-23:
@amitSA Are you still here? 

I finally got around to trying to reproduce what you describe, and couldn't. 

Do you recall where you added the inline comments that caused the problem?

hansu, 2024-09-23:
> I finally got around to trying to reproduce what you describe, and couldn't.
> 
> Do you recall where you added the inline comments that caused the problem?

Just follow the initial issue description. I just tested on 2.9 and the behaviour has not changed.

andypugh, 2024-09-30:
I just tried actually putting the comment on a numeric value line, and whilst the error message returned was cryptic and unhelpful, I didn't see a hang.

````
LINUXCNC - 2.9.3
andypugh@rm-one:~/linuxcnc-dev/src$ geany &
[1] 40874
andypugh@rm-one:~/linuxcnc-dev/src$ linuxcnc
LINUXCNC - 2.9.3
Machine configuration directory is '/home/andypugh/linuxcnc/configs/axis'
Machine configuration file is 'axis.ini'

check_config:
  [JOINT_0]MIN_LIMIT > [AXIS_X]MIN_LIMIT ({-10.0  # Random Comment} > -10.0)

check_config validation failed
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/andypugh/linuxcnc_debug.txt
and
    /home/andypugh/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
````

hansu, 2024-10-06:
Seems to be a Gmoccapy-specific problem. With Axis it works as expected.