LinuxCNC/linuxcnc #997: Debugging 2.9 docs.

jmgardeto, 2020-12-09:
The file /configs/by_interface/mesa/hm2-servo/README, line 3, refers to the HostMot2 section in the integrators manual.

That section no longer exists.

Extensive documentation on the hostmot2 driver now appears in LinuxCNC_Documentation.pdf (11.4 Table HostMot2 Driver), or at http://linuxcnc.org/docs/devel/html/drivers/hostmot2.html

I suggest that line 3 of the above file says:
"See LinuxCNC_Documentation.pdf - 11.4 Mesa HostMot2 Driver section for more information."

andypugh, 2020-12-09:
You should be able to edit the file and make a pull request directly on this web site by opening the source file and clicking the "edit" icon (a pencil)

silopolis, 2022-03-05:
Should we link to stable or devel docs in such situations ?