LinuxCNC/linuxcnc #1000: "ignore-limits"-checkbox in gmoccapy 3.0.9.1 (LinuxCNC 2.8) is not displayed correctly

Hans470, 2020-12-12:
## Here are the steps I follow to reproduce the issue:

 1. Drive machine into a limit switch (or simulate the switch)
 2. Set the checkbox "ignore limits" and turn on again
 3. Jog away (with keyboard) from the limit switch

## This is what I expected to happen:

If the limit switch is not active any longer, the checkbox has to be unchecked.
Turn on and off refreshes the checkbox so it display the correct state.

## This is what happened instead:

But it stays active checked even if you stop jogging while the limit switch is still active. But if you try to turn on again because you think the checkbox is checked: no way. Because it is internal (correctly) disabled, but the display is not updated. Unchecking and checking again leads to the expected behaviour.

#### A second thing:

After having such an event the "ignore limits" checkbox is always clickable in off-state. It should be grayed out, when no limit switches are active. Clicking leads to error "no limit switch is active, command will not be executed!" 

## It worked properly before this:

I know that in axis of v2.7 this was working properly. 

## What to do?

I think this is done by updating the checkbox on the following events:

* turn on/turn off
* rising/falling edge of limit switch

Hans470, 2020-12-13:
Edit: I tested it with LinuxCNC 2.7 and it shows the same behaviour.

gmoccapy, 2020-12-31:
I have pushed an bug fix to 2.8 

https://github.com/LinuxCNC/linuxcnc/commit/71349eb590cc7a69993eabff5cee7645661bed66

Should now be solved with gmoccapy 3.0.10
Hans could you please test and report back?
I tested with several sim and under MINT 19.3

Thanks for reporting.

Norbert

Hans470, 2021-01-03:
Hi Norbert,
thanks for dealing with that.

 ### 1. What works now: 
The checkbox is refreshed when jogged away from the limit switch. :heavy_check_mark: 
### 2. What did not work:
 - Try to start with limit switch enabled. You have no chance to enable the machine unless you press and release the e-stop. That lets the checkbox become clickable. When you enable the checkbox after that and then press and release the e-stop (while limit switch is still active) then the checkbox is still displayed as checked but internally isn't.
- In following scenario: limit-sw became active while moving. You clicked the checkbox and turn on again. You jog a bit but not far enough so the movement stops while limit-sw is still active. The machine turns itself off. Then the checkbox is still displayed as checked but internally isn't (like above). So only unchecking and checking again let you enable the machine.

Hans470, 2021-01-03:
Here a screenrecord for case 1:
[image]
and case 2:
[image]

gmoccapy, 2021-01-05:
I have made some more fixing, Hans, thanks for testing.
Would you please try my last push? You should get gmoccapy 3.1.2.1

I tested as far as possible with my sim configs.

Norbert

Hans470, 2021-01-08:
Thanks Norbert.
2. works now, but 1. unfortunately not (enabling the machine with limits active).
Here a sample config you can test with: [config-testlim.tar.gz](https://github.com/LinuxCNC/linuxcnc/files/5786266/config-testlim.tar.gz). There the limit switch is just hardwired to 'tripped'. So if you manage the machine turning on with that config, you have it ;-)
Additionally a very little little thing worked better in your first fix: Wen you move away from he limit switch, in the moment you leave the switch the checkbox is grayed out but the hook disappears not before you release the jog key.

[image]
Not a problem, but if you wanna have it perfect .. ;-)

gmoccapy, 2021-01-18:
[out-1.ogv.zip](https://github.com/LinuxCNC/linuxcnc/files/5831059/out-1.ogv.zip)

I have looked at your config. It is not poosible to use that one for me. I have no such old hardware to support paralell port ;-) So I added the hardlinked limit switches to my sim config. Please see th attached video, I can switch the machine on without any problem. The uncheck and check again of the limit box is IMHO related to the fact, that the home switches will not change their status, because the are hard wired. Also a problem is, that you did connect both limit switches joint.0.max and joint.0.min at the same time, this would not happen in real machine, wont it?

So IMHO it is working fine. I have tested on my real machine, but that one has no limit switches, just reference switches, so I just triggerd my Tool sensor connected to joint.2.pos-lim-switch-in I was not able to get a undefined state.

Norbert

andypugh, 2021-01-18:
Two signals from one sensor is not all that unusual. It is sometimes easier to have one sensor and two flags / targets for example. 

Especially on parallel port systems it is common to connect all limit and home switches for all axes to a single input pin (to save pins)

So, yes, I think that the system does need to be able to cope with both limits being set, possibly on all axes.

Hans470, 2021-01-19:
> I have looked at your config. It is not poosible to use that one for me. I have no such old hardware to support paralell port ;-) So I added the hardlinked limit switches to my sim config. 

You can still ignore the errors on not found parallel port or remove it from the config. It doesn't matter in this case.
But what you have to do is set ` iocontrol.0.emc-enable-in ` to 1 what you actually not have done.
That's why the problem don't appear at your config. The problem appears only if you start it with a not triggered estop.

> P.S. how did you manage to add the animated puctures?

With `byzanz-record`

PS: You can now directly upload videos if you want to have them embedded...

hansu, 2021-03-17:
@gmoccapy Do you have this issue still in mind?
It is still not possible to enable the machine when you start gmoccapy and a limit switch is high and you don't touch  the estop button.
I removed the parallel port from the minal test config: [testlim.tar.gz](https://github.com/LinuxCNC/linuxcnc/files/6159391/testlim.tar.gz)

Here an idea to fix it: https://github.com/LinuxCNC/linuxcnc/compare/2.8...hansu:gmoccapy-limit-switch
In detail:
`self.widgets.chk_ignore_limits.set_active(True)` 
should not be set, it's just for testing. I only want to make it clickable with
`self.widgets.chk_ignore_limits.set_sensitive(True)`
but that didn't work? Why?

PR #1453 (2022-02-27): gmoccapy: fix limit switch issue (#1000)
This fixes https://github.com/LinuxCNC/linuxcnc/issues/1000#issuecomment-753649684

resolves #1000 

@gmoccapy
closed 2022-02-27 by c-morley