LinuxCNC/linuxcnc #1004: Pocket and Tool mixed

TheThom, 2020-12-17:
https://github.com/LinuxCNC/linuxcnc/blob/83a5cd06440de041e0e7485077447aaae963fa0c/nc_files/remap_lib/python-stdglue/stdglue.py#L242

I had some problemes, when the pocket is not the same as the tool-number.
Changed this line to the following, and it worked.

` emccanon.CHANGE_TOOL_NUMBER(self.current_tool) `

andypugh, 2020-12-17:
@rene-dev Is there any chance you could look at this?

TheThom, 2020-12-17:
Sorry, iam new here and i don`t know how..
I had spend several hours to figure out a solution for the problem with my toolchanger.

Now i want to prevent other users from running in the same issue.
If you need any further informations, dont hesitate to ask

TheThom, 2020-12-29:
I was wrong with my idea, sorry.

The prolem is still there. I have no idea why.

e.g. Sometimes a "emccannon.CHANGE_TOOL_NUMBER(4)"  sets tool 5 in the spindle..

closed 2020-12-29 by TheThom