LinuxCNC/linuxcnc #1006: commit 4952f84 seems to break loading Glade files using hal_gremlin

c-morley, 2020-12-19:
In reference to:

https://forum.linuxcnc.org/gmoccapy/35297-gmoccapy-edit-interface#192460

It seems loading of GLADE files that include hal_gremlin crash the GLADE editor because it can't lind an INI.
if you change line 110 to:
        ini_filename = os.environ.get('INI_FILE_NAME', '/dev/null')

(Though I'm not sure if that breaks the intentions of the commit)
you get past that but then it fails when it tries to read parts of the INI

gmoccapy, 2020-12-23:
I am not able to reproduce this.

It is known, that the glade Editor is buggy. I have compiled mine from source! It is necessary to use glade 3.8.6 editor. Please check if you use that release. And it might be necessary to start the editor from a rip-environment terminal.

Norbert

Hans470, 2020-12-24:
Hmm I just build and installed the glade 3.8.6 editor and the same problem (on buster x64).
But on a wheezy 32-bit system it works well with glade 3.8.0. This system was installed using the official iso and then updated to LinuxCNC 2.8. There was even no need to install glade.

c-morley, 2020-12-24:
Yes in second look the crash seems unrelated to the INI error.
Probably a GLADE editor bug.
I'll close this report.

closed 2020-12-24 by c-morley

Hans470, 2021-01-03:
> It is known, that the glade Editor is buggy. I have compiled mine from source! It is necessary to use glade 3.8.6 editor. Please check if you use that release. And it might be necessary to start the editor from a rip-environment terminal.
> 
> Norbert

In a rip environment I got it started. Thanks for that hint. Even with Glade 3.8.0 is it possible if I start it in a rip environment.