LinuxCNC/linuxcnc #1007: configure script is unable to locate boost::python

ragouel, 2020-12-22:
When building from source on Arch Linux, I get this error:

<pre><code>checking for Python libraries... -lpthread -ldl  -lutil
checking whether the Boost::Python headers are available... yes
checking for boost::python shared library... 
configure: error: boost::python is required to build LinuxCNC</code></pre>

## Here are the steps I follow to reproduce the issue:

 1. patch -Np2 -i libtirpc.patch
 2. ./autogen.sh
 3. ./configure --with-realtime=uspace --without-libmodbus --prefix=/usr --with-python=/usr/bin/python2.7 --enable-non-distributable=yes

*boost* and *python* dependencies are installed.

Namely: 
'intltool'
'python2-yapps2'
'bwidget'
'tkimg'
'tclx'
'python2-gtkglext'

[Arch Linux Package Page](https://aur.archlinux.org/packages/linuxcnc/)
[libtirpc.patch](https://aur.archlinux.org/cgit/aur.git/tree/libtirpc.patch?h=linuxcnc)

andypugh, 2020-12-24:
The configure test is here: https://github.com/LinuxCNC/linuxcnc/blob/43ac9730f4d6be679bd28292d88fdfcdb8204907/src/configure.ac#L92

Do you have the specified files in the specified place (I wonder if this is a Python versioning issue?)

lasmahei, 2021-01-02:
I have exactly the same issue. Would be great of someone finds the cause to be able to install linuxcnc again.

andypugh, 2021-01-02:
When you say "Install" do you mean "Compile"?

If you run the tests above manually, what do you find?

UweSauter, 2021-01-30:
Trying to help here though I have no knowledge regarding Autotools.

The `boost` package on Arch installs the following files that have `python` in their path component:
```
/usr/include/boost/mpi/python.hpp
/usr/include/boost/mpi/python/
/usr/include/boost/mpi/python/config.hpp
/usr/include/boost/mpi/python/serialize.hpp
/usr/include/boost/mpi/python/skeleton_and_content.hpp
/usr/include/boost/parameter/aux_/python/
/usr/include/boost/parameter/aux_/python/invoker.hpp
/usr/include/boost/parameter/aux_/python/invoker_iterate.hpp
/usr/include/boost/parameter/python.hpp
/usr/include/boost/python.hpp
/usr/include/boost/python/
/usr/include/boost/python/arg_from_python.hpp
/usr/include/boost/python/args.hpp
/usr/include/boost/python/args_fwd.hpp
/usr/include/boost/python/back_reference.hpp
/usr/include/boost/python/base_type_traits.hpp
/usr/include/boost/python/bases.hpp
/usr/include/boost/python/borrowed.hpp
/usr/include/boost/python/call.hpp
/usr/include/boost/python/call_method.hpp
/usr/include/boost/python/cast.hpp
/usr/include/boost/python/class.hpp
/usr/include/boost/python/class_fwd.hpp
/usr/include/boost/python/converter/
/usr/include/boost/python/converter/arg_from_python.hpp
/usr/include/boost/python/converter/arg_to_python.hpp
/usr/include/boost/python/converter/arg_to_python_base.hpp
/usr/include/boost/python/converter/as_to_python_function.hpp
/usr/include/boost/python/converter/builtin_converters.hpp
/usr/include/boost/python/converter/constructor_function.hpp
/usr/include/boost/python/converter/context_result_converter.hpp
/usr/include/boost/python/converter/convertible_function.hpp
/usr/include/boost/python/converter/from_python.hpp
/usr/include/boost/python/converter/implicit.hpp
/usr/include/boost/python/converter/obj_mgr_arg_from_python.hpp
/usr/include/boost/python/converter/object_manager.hpp
/usr/include/boost/python/converter/pointer_type_id.hpp
/usr/include/boost/python/converter/pyobject_traits.hpp
/usr/include/boost/python/converter/pyobject_type.hpp
/usr/include/boost/python/converter/pytype_function.hpp
/usr/include/boost/python/converter/pytype_object_mgr_traits.hpp
/usr/include/boost/python/converter/registered.hpp
/usr/include/boost/python/converter/registered_pointee.hpp
/usr/include/boost/python/converter/registrations.hpp
/usr/include/boost/python/converter/registry.hpp
/usr/include/boost/python/converter/return_from_python.hpp
/usr/include/boost/python/converter/rvalue_from_python_data.hpp
/usr/include/boost/python/converter/shared_ptr_deleter.hpp
/usr/include/boost/python/converter/shared_ptr_from_python.hpp
/usr/include/boost/python/converter/shared_ptr_to_python.hpp
/usr/include/boost/python/converter/to_python_function_type.hpp
/usr/include/boost/python/copy_const_reference.hpp
/usr/include/boost/python/copy_non_const_reference.hpp
/usr/include/boost/python/data_members.hpp
/usr/include/boost/python/def.hpp
/usr/include/boost/python/def_visitor.hpp
/usr/include/boost/python/default_call_policies.hpp
/usr/include/boost/python/detail/
/usr/include/boost/python/detail/aix_init_module.hpp
/usr/include/boost/python/detail/api_placeholder.hpp
/usr/include/boost/python/detail/borrowed_ptr.hpp
/usr/include/boost/python/detail/caller.hpp
/usr/include/boost/python/detail/config.hpp
/usr/include/boost/python/detail/construct.hpp
/usr/include/boost/python/detail/convertible.hpp
/usr/include/boost/python/detail/copy_ctor_mutates_rhs.hpp
[... 181 lines cut ...]
/usr/lib/libboost_mpi_python39.a
/usr/lib/libboost_python39.a
```
The one header the test wants to use (`boost/python/object_fwd.hpp`) is installed in `/usr/include/boost/python/`.

There's also a second package in Arch Linux: `boost-libs` which provides the following files with `python` in their path:
```
/usr/lib/boost-python3.9/
/usr/lib/libboost_mpi_python39.so
/usr/lib/libboost_mpi_python39.so.1.75.0
/usr/lib/libboost_python3.so
/usr/lib/libboost_python39.so
/usr/lib/libboost_python39.so.1.75.0
/usr/lib/python3.9/
/usr/lib/python3.9/site-packages/
/usr/lib/python3.9/site-packages/boost/
/usr/lib/python3.9/site-packages/boost/__init__.py
/usr/lib/python3.9/site-packages/boost/__pycache__/
/usr/lib/python3.9/site-packages/boost/__pycache__/__init__.cpython-39.opt-1.pyc
/usr/lib/python3.9/site-packages/boost/__pycache__/__init__.cpython-39.opt-2.pyc
/usr/lib/python3.9/site-packages/boost/__pycache__/__init__.cpython-39.pyc
/usr/lib/python3.9/site-packages/boost/mpi.so
```

If you can provide instructions how to run that particular test I'll gladly provide the output.

UweSauter, 2021-02-04:
Improving above test it seems as if there is a missing include path for Python:

```
$ sudo find / -xdev -name pyconfig.h
/usr/include/python2.7/pyconfig.h
/usr/include/python3.9/pyconfig.h
```
```
$ cat test.cpp 
#include <boost/python/object_fwd.hpp>
boost::python::object *test_object;

int main() {
}

$ LANG=C g++ -o test -I/usr/include/python3.9/ test.cpp
$ echo $?
0
```
But I still don't know if this is the cause for the Autoconf test failing.

UweSauter, 2021-02-04:
So, I found out that Autoconf creates a `config.log`. The relevant part for this error is:
```
configure:10022: checking whether the Boost::Python headers are available
configure:10042: g++ -c -g -O2 -I/usr/include/python2.7  conftest.cpp >&5
configure:10042: $? = 0
configure:10050: result: yes
configure:10068: checking for boost::python shared library
configure:10085: g++ -o conftest -g -O2 -I/usr/include/python2.7   conftest.cpp  -lXinerama -ltirpc  -lpython2.7 -lboost_python >&5
/usr/bin/ld: cannot find -lboost_python
collect2: error: ld returned 1 exit status
configure:10085: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "LinuxCNC"
| #define PACKAGE_TARNAME "linuxcnc"
| #define PACKAGE_VERSION "2.8.1"
| #define PACKAGE_STRING "LinuxCNC 2.8.1"
| #define PACKAGE_BUGREPORT "emc-developers@lists.sourceforge.net"
| #define PACKAGE_URL ""
| #define HAVE_STDIO_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_STRING_H 1
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_STRINGS_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_UNISTD_H 1
| #define STDC_HEADERS 1
| #define HAVE_CXX11 1
| #define HAVE_LIBTIRPC 1
| #define RTAPI_USPACE /**/
| #define RTDIR "/home/uwe/AUR/linuxcnc/src/linuxcnc-2.8.1/src"
| #define HIDRAW_H_USABLE /**/
| #define HAVE_LIBUSB10 yes
| #define MODULE_EXT ".so"
| #define EMC2_DEFAULT_NMLFILE "/usr/share/linuxcnc/linuxcnc.nml"
| #define EMC2_DEFAULT_TOOLTABLE ""
| #define EMC2_HOME "/usr"
| #define EMC2_BIN_DIR "/usr/bin"
| #define EMC2_TCL_DIR "/usr/lib/tcltk/linuxcnc"
| #define EMC2_HELP_DIR "/usr/share/doc/linuxcnc"
| #define EMC2_RTLIB_DIR "/usr/lib/linuxcnc/modules"
| #define EMC2_LANG_DIR "/usr/lib/tcltk/linuxcnc/msgs"
| #define EMC2_PO_DIR "/usr/share/locale"
| #define EMC2_CONFIG_DIR ""
| #define EMC2_NCFILES_DIR "/usr/share/linuxcnc/ncfiles"
| #define EMC2_IMAGE_DIR "/usr/share/linuxcnc"
| #define HAVE_SYS_IO_H 1
| #define HAVE_SYS_WAIT_H 1
| #define HAVE_SEMTIMEDOP 1
| #define LIBDL $LIBDL
| #define HAVE_CLOCK_NANOSLEEP 1
| #define HAVE_X11_EXTENSIONS_XINERAMA_H 1
| #define HAVE_LIBXINERAMA 1
| #define HAVE_READLINE_READLINE_H 1
| #define HAVE_READLINE_HISTORY_H 1
| #define HAVE_READLINE /**/
| #define HAVE_LOCALE_H 1
| #define HAVE_SETLOCALE 1
| #define HAVE_LIBINTL_H 1
| /* end confdefs.h.  */
| #include <boost/python/object_fwd.hpp>
| int
| main (void)
| {
| boost::python::object *test_object;
|   ;
|   return 0;
| }
```
In the comments on the Arch AUR package page some other user reported that he was able to solve the issue by installing the `boost-python2` package from the AUR. I can confirm that this allows LinuxCNC to build.