LinuxCNC/linuxcnc #1012: python remap: yield doesn't return if there was an error while waiting.

c-morley, 2020-12-28:
yield is used in remaps to wait for linuxcnc to execute some code. ie wait for probe to finish.
This works fine if the executed code was successful.
If it wasn't, linuxcnc does not return to the yield point, making error handling impossible.
in the case of the tool_probe_m6 remap, the system will be stuck in G90 and possibly the wrong units (G21/20).

It's not clear to me from docs, if this is a short coming of remap or just a bug.

There is a remap that shows the problem in linuxcnc:nc_files/remap_lib/python-stdglue/tool_probe_m6
If the probing ends before actually touching off, the remap is abandoned in a weird state.

interp_python.cc  pycall() seems to control the python generator/yield process.

My c++ is pretty non-existent.
If anyone could assist - i'll do my best to work this out.

c-morley, 2020-12-28:
I haven't tested actually 2.8 - only master

rene-dev, 2020-12-28:
could it be related to https://github.com/LinuxCNC/linuxcnc/issues/588?

c-morley, 2020-12-28:
Yes I'm sure it is. This was a pure python remap but the process is surely similar.
I'll see if I can confirm that.
Thanks for pointing it out.
should I close this one then?

c-morley, 2020-12-29:
This is a sample with DEBUG = 0x7FFFFFFF (all)
and in the ini {PYTHON} LOG_LEVEL=1000

using the new remap tool_probe_m6 in std_glue (master)
`
**** issued t1m6 in MDI

Issuing EMC_TASK_PLAN_EXECUTE -- 	 (  +509,+280,   +12,t1m6\n,)
emcTaskPlanLevel() returned 0
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=24,type=EMC_TOOL_PREPARE}) : list_size=1, line_number=0
is_callable(oword.tool_probe_m6) = FALSE
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=168,type=EMC_TRAJ_SET_OFFSET}) : list_size=2, line_number=0
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=208,type=EMC_TRAJ_LINEAR_MOVE}) : list_size=3, line_number=0
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=24,type=EMC_TOOL_LOAD}) : list_size=4, line_number=0
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=208,type=EMC_TRAJ_LINEAR_MOVE}) : list_size=5, line_number=0
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=208,type=EMC_TRAJ_LINEAR_MOVE}) : list_size=6, line_number=0
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=96,type=EMC_TRAJ_CLEAR_PROBE_TRIPPED_FLAG}) : list_size=7, line_number=0
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=208,type=EMC_TRAJ_PROBE}) : list_size=8, line_number=0
emcTaskPlanExecute(t1m6
) returned 2
emcTaskPlanLevel() returned 1
emcTaskPlanSetWait() called
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=24,type=EMC_TASK_PLAN_SYNCH}) : list_size=9, line_number=0
NML_INTERP_LIST(0x90bde0)::get(): {size=24, type=EMC_TOOL_PREPARE}, list_size=8
emcTaskPlanLevel() returned 1
Issuing EMC_TOOL_PREPARE -- 	 ( +1104,+24,    +0,    +1,)
NML_INTERP_LIST(0x90bde0)::get(): {size=168, type=EMC_TRAJ_SET_OFFSET}, list_size=7
emcTaskPlanLevel() returned 1
Issuing EMC_TRAJ_SET_OFFSET -- 	 (  +223,+168,    +0,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,)
NML_INTERP_LIST(0x90bde0)::get(): {size=208, type=EMC_TRAJ_LINEAR_MOVE}, list_size=6
emcTaskPlanLevel() returned 1
Issuing EMC_TRAJ_LINEAR_MOVE -- 	 (  +220,+208,    +0,10.000000,8.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,    +4,6.397584,6.397584,63.975838,    +0,    -1,)
NML_INTERP_LIST(0x90bde0)::get(): {size=24, type=EMC_TOOL_LOAD}, list_size=5
emcTaskPlanLevel() returned 1
Issuing EMC_TOOL_LOAD -- 	 ( +1105,+24,    +0,)
NML_INTERP_LIST(0x90bde0)::get(): {size=208, type=EMC_TRAJ_LINEAR_MOVE}, list_size=4
emcTaskPlanLevel() returned 1

*** waits for manual tool change acknowledge

Issuing EMC_TRAJ_LINEAR_MOVE -- 	 (  +220,+208,    +0,-10.000000,-10.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,    +1,7.064806,7.064806,70.648061,    +0,    -1,)
NML_INTERP_LIST(0x90bde0)::get(): {size=208, type=EMC_TRAJ_LINEAR_MOVE}, list_size=3
emcTaskPlanLevel() returned 1
Issuing EMC_TRAJ_LINEAR_MOVE -- 	 (  +220,+208,    +0,-10.000000,-10.000000,0.110000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,    +1,5.000000,5.000000,50.000000,    +0,    -1,)
NML_INTERP_LIST(0x90bde0)::get(): {size=96, type=EMC_TRAJ_CLEAR_PROBE_TRIPPED_FLAG}, list_size=2
emcTaskPlanLevel() returned 1
Issuing EMC_TRAJ_CLEAR_PROBE_TRIPPED_FLAG -- 	 (  +228,+96,    +0,)
NML_INTERP_LIST(0x90bde0)::get(): {size=208, type=EMC_TRAJ_PROBE}, list_size=1
emcTaskPlanLevel() returned 1
Issuing EMC_TRAJ_PROBE -- 	 (  +229,+208,    +0,-10.000000,-10.000000,-6.890000,0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,    +5,1.000000,5.000000,50.000000,\000,)
NML_INTERP_LIST(0x90bde0)::get(): {size=24, type=EMC_TASK_PLAN_SYNCH}, list_size=0
emcTaskPlanLevel() returned 1

**** probing --- then fails

G38.2 move finished without making contact.
NML_INTERP_LIST(0x90bde0)::clear(): discarding 0 items
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=24,type=EMC_TASK_PLAN_SYNCH}) : list_size=1, line_number=0
NML_INTERP_LIST(0x681aa0)::clear(): discarding 0 items
NML_INTERP_LIST(0x90bde0)::clear(): discarding 1 items
NML_INTERP_LIST(0x90bde0)::append(nml_msg_ptr{size=24,type=EMC_TASK_PLAN_SYNCH}) : list_size=1, line_number=0
task: main loop took 0.019002 seconds
NML_INTERP_LIST(0x90bde0)::get(): {size=24, type=EMC_TASK_PLAN_SYNCH}, list_size=0
emcTaskPlanLevel() returned 0
Issuing EMC_TASK_PLAN_SYNCH -- 	 (  +516,+24,    +0,)
emcTaskPlanSynch() returned 0

*** process finished
`