LinuxCNC/linuxcnc #1018: linux-image-4.19.71-rt24-v7l+ package on Raspberry Pi has mistype in control file

antage, 2020-12-30:
After following instructions at [manual](http://linuxcnc.org/docs/2.8/html/getting-started/getting-linuxcnc.html#cha:Installing-RTAI) I try to install kernel but get error:

```
Unpacking linux-image-4.19.71-rt24-v7l+ (4.19.71-rt24-v7l+-4) ...
dpkg: error processing archive /var/cache/apt/archives/linux-image-4.19.71-rt24-v7l+_4.19.71-rt24-v7l+-4_armhf.deb (--unpack):
 trying to overwrite '/boot/bcm2708-rpi-b-plus.dtb', which is also in package raspberrypi-kernel 1.20201201-1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
```

It's because the deb-package control file has line:
```
Conflicts: raspberrpi-kernel
```

but it should be:

```
Conflicts: raspberrypi-kernel
```

antage, 2020-12-30:
And it's better if we'll use `Replaces` instead of `Conflicts`.

andypugh, 2021-01-02:
I have tried to correct this, and the kernel does now install without errors. However after doing so my Pi does not boot from the new kernel. 
(I have tried several times from a freshly-made SD card) 
Can you try this .deb and see if you get the same result? 
https://www.linuxcnc.org/temp/linux-image-4.19.71-rt24-v7l+_4.19.71-rt24-v7l+-5_armhf.deb

antage, 2021-01-02:
I had tried to remove `raspberrypi-kernel` package and install `linux-image-4.19.71-rt24-v7l+` (without the fix) and RPi doesn't boot too.
But with the kernel from source (I use [this manual](https://gnipsel.com/linuxcnc/uspace/rpi4-rt.html)) all works fine.

antage, 2021-01-02:
I take a look inside deb-package and I see the kernel is installed as `/boot/vmlinuz-4.19.71-rt24-v7l+` but `/boot/start.elf` loads `/boot/kernel7l.img` (on RPi4) or takes kernel filename from `/boot/config.txt:kernel=`.
If I'm right after `linux-image-4.19.71-rt24-v7l+` has been installed RPi is trying to load the old kernel with dtb and dtbo files from 4.19-rt24 kernel.

andypugh, 2021-01-03:
Can you suggest a solution? It is clear to me that you know a lot more about this than I do. 
I have been working on LinuxCNC for a long time, but making kernel packages is new to me, and something I was forced to pick up when I accepted the job of release manager for 2.8. 
We moved the Pi kernel sources that we use here:
https://github.com/LinuxCNC/raspberrypi-linux
(Note that I haven't actually pushed the change to that branch yet, but this is the commit that introduced the issue:
https://github.com/LinuxCNC/raspberrypi-linux/commit/f2a59ba90cd4185a6867a6a95714966b91e35305 )

antage, 2021-01-03:
@andypugh Ok, I'll try to find a solution.

antage, 2021-01-04:
I'm sorry I mix all in this issue but could you take a look at https://github.com/raspberrypi/linux/pull/4046 ?
Without this patch 4.19.y-rt has no USB on RPi4.
It's merged to [raspberrypi/linux](https://github.com/raspberrypi/linux)@rpi-4.19.y-rt [already](https://github.com/raspberrypi/linux/commits/rpi-4.19.y-rt).

andypugh, 2021-01-04:
I am confused, I have been running the LinuxCNC version of the 4.19 kernel for months on my test Pi4 and have had no USB problems.

antage, 2021-01-04:
I'm unsure but I think it exists different hardware revisions. Newer uses VIA VL805 as USB3.0 chipset:

[image]

What's your USB chipset?

antage, 2021-01-04:
My VL805 firmware version:

```
$ sudo rpi-eeprom-update 
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Fri 11 Dec 2020 11:15:17 AM UTC (1607685317)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

antage, 2021-01-04:
More [reports](https://forum.linuxcnc.org/9-installing-linuxcnc/40466-linuxcnc-2-8-0-pi4-img-no-usb-on-raspberry-pi-4-8gb-2gb) about USB issue.

antage, 2021-01-05:
I found HW revision 1.4 has no SPI EEPROM chip with VL805 firmware so CPU has to load the firmware (the patch does it).
HW revisions <1.4 loads VL805 firmware from SPI EEPROM and it doesn't require additional code in Linux.

[image]

andypugh, 2021-01-06:
I tried to transfer this to the linuxcnc/raspberry-linux repository, but for some reason Github won't allow that.

antage, 2021-01-06:
I see all changes have been committed. Thanks!

closed 2021-01-06 by antage

andypugh, 2021-01-11:
Did you test the new kernel? The reports I see on the forum suggest that the USB issue is _not_ fixed.

antage, 2021-01-12:
I've installed the new kernel from linuxcnc apt-repo and USB works on my RPi.
But I notice you didn't increment the package version. It may be a problem if OS was installed from linuxcnc image or the package was installed early.

andypugh, 2021-01-12:
There might be an additional problem with the 8GB:
https://forum.linuxcnc.org/9-installing-linuxcnc/40466-linuxcnc-2-8-0-pi4-img-no-usb-on-raspberry-pi-4-8gb-2gb?start=10#194805